particle-iot/device-os #1176: [Electron] Application/System thread not blocking on calls to modem with SYSTEM_THREAD(ENABLED)

technobly, 2016-11-15:
SYSTEM_THREAD(ENABLED) is not blocking on calls to the modem, so the application thread doesn't wait for the modem to finish before going to the next instruction in user code.  Thus Cellular.on() and SLEEP_MODE_SOFTPOWEROFF will not work properly with threading enabled currently.

## Example of non working code
```c++
SYSTEM_MODE(SEMI_AUTOMATIC);
SYSTEM_THREAD(ENABLED);

void setup() {
}

void loop() {
  // we need to turn on the modem in SEMI_AUTOMATIC mode first, 
  // so that AT commands can be sent to turn it off
  Cellular.on();
  // system will not wait for power on AT command sequence

  System.sleep(SLEEP_MODE_DEEP, 15); 
  // system will not wait for power off AT command sequence
}

// LOGS
0000000002 [system] main.cpp:269, system_power_management_init(): INFO: Power Management Initializing.
0000000014 [system] main.cpp:282, system_power_management_init(): INFO: State of Charge: 68.17 %
0000000017 [system] main.cpp:283, system_power_management_init(): INFO: Battery Voltage: 3.94V
0000000025 [system] main.cpp:579, app_setup_and_loop(): TRACE: Hello from Particle!
0000000033 [system] main.cpp:581, app_setup_and_loop(): INFO: Device 410061000b51343334363138 started
0000000041 [system] main.cpp:587, app_setup_and_loop(): TRACE: Last reset reason: 30 (data: 0x00)
0000000052 [usb.composite] usbd_composite.c:187, USBD_Composite_DeInit(): TRACE: DeInitializing (0x0)
0000000061 [comm] communication_dynalib.cpp:56, spark_protocol_instance(): TRACE: UDP enabled 1
0000000067 [comm] communication_dynalib.cpp:31, create_protocol(): TRACE: creating DTLS protocol
0000000078 [comm] dtls_protocol.cpp:47, particle::protocol::DTLSProtocol::init(): INFO: channel inited
0000000313 [usb.composite] usbd_composite.c:403, USBD_Build_CfgDesc(): TRACE: Built configuration (0x1): 84 bytes, 5 total interfaces, 3 active
0000000315 [usb.composite] usbd_composite.c:403, USBD_Build_CfgDesc(): TRACE: Built configuration (0x1): 84 bytes, 5 total interfaces, 3 active
0000000316 [usb.composite] usbd_composite.c:164, USBD_Composite_Init(): TRACE: Initializing (0x1)
[ ElectronSe�

Here is where it is trying to turn on ^, system blows right past and goes to sleep
```

As a work around you can break out the commands separately and add delays after the commands.  The delays need to be longer than the commands could possibly take.  10 seconds after Cellular.on is typically enough, but there is an edge case where the modem doesn't respond to the power pin being toggled and needs to then be reset after 10 seconds of trying to power it up.  So in that case it could take more like 15 - 20 seconds max to power on.  Here in code I've just used 10 seconds though:

## Work around example
```c++
SYSTEM_MODE(SEMI_AUTOMATIC);
SYSTEM_THREAD(ENABLED);

void setup() {
}

void loop() {
  // we need to turn on the modem in SEMI_AUTOMATIC mode first, 
  // so that AT commands can be sent to turn it off
  Cellular.on();
  delay(10000); // this can really be up to 20 seconds worst case

  Cellular.command("AT+CPWROFF\r\n");
  delay(2000);

  FuelGauge().sleep();
  delay(2000);

  System.sleep(SLEEP_MODE_DEEP, 15); // just a normal DEEP sleep can be called instead of SOFTPOWEROFF now that we already turned off the modem and fuel gauge.
}

// LOGS
0000000002 [system] main.cpp:269, system_power_management_init(): INFO: Power Management Initializing.
0000000014 [system] main.cpp:282, system_power_management_init(): INFO: State of Charge: 68.17 %
0000000017 [system] main.cpp:283, system_power_management_init(): INFO: Battery Voltage: 3.94V
0000000025 [system] main.cpp:579, app_setup_and_loop(): TRACE: Hello from Particle!
0000000033 [system] main.cpp:581, app_setup_and_loop(): INFO: Device 410061000b51343334363138 started
0000000041 [system] main.cpp:587, app_setup_and_loop(): TRACE: Last reset reason: 30 (data: 0x00)
0000000052 [usb.composite] usbd_composite.c:187, USBD_Composite_DeInit(): TRACE: DeInitializing (0x0)
0000000061 [comm] communication_dynalib.cpp:56, spark_protocol_instance(): TRACE: UDP enabled 1
0000000067 [comm] communication_dynalib.cpp:31, create_protocol(): TRACE: creating DTLS protocol
0000000078 [comm] dtls_protocol.cpp:47, particle::protocol::DTLSProtocol::init(): INFO: channel inited
0000000321 [usb.composite] usbd_composite.c:403, USBD_Build_CfgDesc(): TRACE: Built configuration (0x1): 84 bytes, 5 total interfaces, 3 active
0000000322 [usb.composite] usbd_composite.c:403, USBD_Build_CfgDesc(): TRACE: Built configuration (0x1): 84 bytes, 5 total interfaces, 3 active
0000000324 [usb.composite] usbd_composite.c:164, USBD_Composite_Init(): TRACE: Initializing (0x1)
[ ElectronSerialPipe::begin ] = = = = = = = =

[ Modem::powerOn ] = = = = = = = = = = = = = =
     2.858 AT send       4 "AT\r\n"
     4.178 AT send       4 "AT\r\n"
     5.498 AT send       4 "AT\r\n"
     5.508 AT read UNK   3 "AT\r"
     5.518 AT read OK    6 "\r\nOK\r\n"
     5.518 AT send       7 "AT E0\r\n"
     5.529 AT read UNK   6 "AT E0\r"
     5.539 AT read OK    6 "\r\nOK\r\n"
     5.539 AT send      11 "AT+CMEE=2\r\n"
     5.550 AT read OK    6 "\r\nOK\r\n"
     5.550 AT send      19 "AT+CMER=1,0,0,2,1\r\n"
     5.562 AT read OK    6 "\r\nOK\r\n"
     5.562 AT send      15 "AT+IPR=115200\r\n"
     5.574 AT read OK    6 "\r\nOK\r\n"
     5.674 AT send      10 "AT+CPIN?\r\n"
     5.684 AT read  +   16 "\r\n+CPIN: READY\r\n"
     5.694 AT read OK    6 "\r\nOK\r\n"
    12.545 AT send      12 "AT+CPWROFF\r\n"
    12.696 AT read OK    6 "\r\nOK\r\n"
```

## Proposed solution

System/Application calls to the modem should block until complete. `delay()` blocks currently in threaded code.

---

Completeness:
- [x] Minimum test case added
- [x] Device, system and user firmware versions stated (v0.6.0)
- [x] Particle confirmed

robynjayqueerie, 2016-12-02:
If I could add a comment here. I don't think you want the call to the modem to block necessarily because a good RTOS should keep going and give a flag or semaphore when the task is finished.
Another comment, as this thread has come from the sleep mode being activated before the modem really shuts down. In my opinion sleep mode should be blocked until the system thread has executed everything it has to do. That would make more sense.
If I am running several tasks in parallel I don't want a call to Cellular.off() to stop everything else working for 10 or more seconds. There could be sensors etc to read. What I really need is the information on when the modem is actually off. Also if I make a call to sleep, it should not be activated if the system thread is still trying to do something. Only the system thread is aware of what it is doing and what it has to finish before the cpu stops operating.

technobly, 2016-12-02:
Good points @robynjayqueerie 

System.sleep... does currently wait for some system activity to complete, but not calls to the modem.  Not blocking on calls to the modem can be problematic if you have no way to be notified when that process completes.  There are definitely more than a couple ways to solve this.  I'll add another to the original proposal, a Futures API.

m-mcgowan, 2016-12-02:
Totally, when I designed the threading I had in mind using futures but dropped it from the first revision to keep the spec minimal. 

We can use the same futures API that @sergeuz has added for `Particle.publish()`.  It also elegantly solves the problem of allowing user code to determine if functions that return values should be executed synchronously or asynchronously. The default will be asynchronous unless the caller attempts to check for the result of the function.  For `void` functions, the caller will have to explicitly wait on the future to get synchronous behavior.

PR #1259 (2017-05-04): Make System.sleep(mode, seconds) a synchronous operation
This PR fixes #1176 by making `System.sleep(mode, seconds)` a synchronous operation in multithreaded firmware. This allows to ensure that the device is in a well-defined state before entering sleep mode.

---

Doneness:

- [X] Contributor has signed CLA
- [X] Problem and Solution clearly stated
- [ ] Code peer reviewed
- [ ] API tests compiled
- [ ] Run unit/integration/application tests on device
- [ ] Add documentation
- [ ] Add to CHANGELOG.md after merging (add links to docs and issues)
closed 2017-05-04 by technobly