particle-iot/device-os #1185: Logging level LOG_LEVEL_NONE outputs log

hramadan, 2016-11-28:
Log() should not output anything when logging level is LOG_LEVEL_NONE

---------------------------------------
SerialLogHandler logHandler;

void setup() {
}
void loop() {
    Log(LOG_LEVEL_TRACE, "This is a trace message");
    Log(LOG_LEVEL_INFO, "This is an info message");
    Log(LOG_LEVEL_WARN, "This is a warn message");
    Log(LOG_LEVEL_ERROR, "This is an error message");
    Log(LOG_LEVEL_NONE, "This is none message");
    delay(1000L);
}
---------------------------------------
Outputs:

0000552305 [app] INFO: This is an info message
0000552305 [app] WARN: This is a warn message
0000552305 [app] ERROR: This is an error message
0000552305 [app] PANIC: This is none message

Log message is unexpected as is the the "PANIC" description. 

Documentation states:

LOG_LEVEL_NONE : special value that can be used to disable logging of any messages

Firmware v0.6.0, PHOTON.

---

Completeness:
- [X] Minimum test case added
- [X] Device, system and user firmware versions stated
- [ ] Particle confirmed

sergeuz, 2016-11-28:
Thanks for your report, @hramadan. `LOG_LEVEL_NONE` (as well as `LOG_LEVEL_ALL`) is a "special value" in the sense that it can be used for filtering, where you can disable logging of particular messages, rather than logging itself:

```cpp
SerialLogHandler logHandler(LOG_LEVEL_ALL, { // Log all messages by default
    { "system", LOG_LEVEL_NONE } // Filter out all messages generated by the system module
});
```

Looks like we need to improve our documentation in this aspect.

hramadan, 2016-11-28:
@sergeuz thanks for this. 

I will close this ticket as it looks to be documentation rather than code!

closed 2016-11-28 by hramadan