particle-iot/device-os #1197: Local building with v0.6.0 Platform 6 Prod 6 causes problems with SMT32_USB_OTG

bkobkobko, 2016-12-04:
In thiis [forum post](http://community.particle.io/t/solved-stm32-usb-otg-driver-error-on-v0-6-0/26814/3 ) several folks complain that building locally for platform 6, product ID 6 causes problems with the STM32_USB_OTG driver such that the build does not complete.  A suggested alternative was to lower the gcc optimization level via `#pragma GCC optimize ("O1")` but that has a number of downsides since this file is from ST Mirco and the code size increase can make the firmware no longer fit.

```
Building firmware for Production Photon, platform ID: 6, product ID: 6
src/main.c: In function 'main':
MCU/STM32F2xx/STM32_USB_OTG_Driver/src/usb_core.c:301:27: error: iteration 6 invokes undefined behavior [-Werror=aggressive-loop-optimizations]
MCU/STM32F2xx/STM32_USB_OTG_Driver/src/usb_core.c:299:3: note: containing loop
MCU/STM32F2xx/STM32_USB_OTG_Driver/src/usb_core.c:307:25: error: iteration 6 invokes undefined behavior [-Werror=aggressive-loop-optimizations]
MCU/STM32F2xx/STM32_USB_OTG_Driver/src/usb_core.c:305:3: note: containing loop
lto1: all warnings being treated as errors
lto-wrapper: arm-none-eabi-g++ returned 1 exit status
/home/ubuntu/bin/gcc-arm-embedded/gcc-arm-none-eabi-4_9-2015q3/bin/../lib/gcc/arm-none-eabi/4.9.3/../../../../arm-none-eabi/bin/ld: lto-wrapper failed
collect2: error: ld returned 1 exit status
make[1]: *** [../build/target/bootloader/platform-6-lto/bootloader.elf] Error 1
make[1]: Target `all' not remade because of errors.
make: *** [bootloader] Error 2
   text    data     bss     dec     hex filename
   4524       8    1460    5992    1768 /home/ubuntu/workspace/bin/firmware.elf
make: Target `all' not remade because of errors.
```

---

Completeness:
- [X] Minimum test case added
- [X] Device, system and user firmware versions stated
- [ ] Particle confirmed

sergeuz, 2016-12-04:
Should be fixed by this PR: https://github.com/spark/firmware/pull/1179 (already merged to `develop`).

technobly, 2017-02-03:
Appears to be shown as resolved in the community, fixed in 0.6.1-rc.1.  Thanks @bkobkobko !

closed 2017-02-03 by technobly