particle-iot/device-os #1198: 1. Download and Install Dependencies, things not clear

versamodule, 2016-12-10:
I wanted to compile locally on my Win7 PC. When I try following the steps here, 

https://github.com/spark/firmware/blob/latest/docs/dependencies.md#1-download-and-install-dependencies

Things are ok to understand until get to "6. Command line tools", These are my issues with it not being clear. 

#1)  "On Windows, you'll need to install MinGW and have it in your path to make available some of the typical *nix command line tools. " 

Ok, so where is the link? All the other steps were given one. Thought no big deal would google it.
http://mingw.org/
I click on downloads and then have no idea which one to download. It would be nice if this was clearer. 

#2) The tool  crc32  is also needed:

Ok, but again where do we get this ?

technobly, 2016-12-11:
Please try the installer: https://binaries.particle.io/cli/installer/windows/ParticleCLISetup.exe

versamodule, 2016-12-11:
Thank you, but I am not sure what the point of it was for. 
I did run it, but still have the crc32 too not found issue. 

FWIW, I did like that installer, its it not possible to make it do everything ?

technobly, 2016-12-11:
Honestly I thought the crc32 might have been in the installer. I guess we'll have to add that!  I'll try to help you out tomorrow on this if you haven't figured out a source for the crc32 by then.

versamodule, 2016-12-11:
That would be fine, and appreciated. I posted youtube vid many years ago how to do it, 
It's now obsolete and does not  work, so I wanted a modern and updated one.

versamodule, 2016-12-12:
That is a really nice tool!! Is it publicly available, and if not will it be ?
I guess what I am asking, is if there are any new updates to it in the future, would I use the same download link ?

technobly, 2016-12-12:
@versamodule two solutions:

Build them yourself:
1) https://github.com/rickkas7/crc32_and_xxd

Or trust that these exe's are virus/trojan free (I did an online virus scan and crc32.exe was 0/56 and xxd.exe was 1/56)
2) https://dl.dropboxusercontent.com/u/41117656/crc32_and_xxd_for_windows.zip

You still need to put the crc32.exe and xxd.exe somewhere and add that to your PATH.

Updates to Particle CLI Installer will be at the same URL.  Unfortunately I don't see any way we are currently announcing updates to it: https://docs.particle.io/guide/tools-and-features/cli/photon/

closed 2016-12-12 by versamodule