particle-iot/device-os #1201: Publish with NO_ACK missing ttl field?

kennethlimcp, 2016-12-12:
A user tried doing `Particle.publish("t", temperature, ttl, PRIVATE, NO_ACK);`: http://community.particle.io/t/problems-compiling-particle-publish-with-no-ack-on-electron-solved/27924

as per documented here: https://docs.particle.io/reference/firmware/electron/#particle-publish- 

but it seems like it's not working.

Test case on v0.6.0:

```
void setup()
{
  Particle.publish("t", 10, 10, PRIVATE, NO_ACK);
}
```

---

Completeness:
- [x] Minimum test case added
- [x] Device, system and user firmware versions stated
- [ ] Particle confirmed

m-mcgowan, 2016-12-13:
Thanks for the report. 

> but it seems like it's not working.

We need a bit more to do on ;-) Please explain the observed behaviour vs expected behaviour. Thanks!

kennethlimcp, 2016-12-13:
opps..

```
test.cpp:12:50: error: no matching function for call to 'CloudClass::publish(const char [2], const char [3], int, const PublishFlag&, const PublishFlag&)'
 
                                                  ^

test.cpp:12:50: note: candidates are:
In file included from ../wiring/inc/spark_wiring.h:48:0,
                 from ./inc/application.h:36,
                 from test.cpp:2:
../wiring/inc/spark_wiring_cloud.h:222:17: note: bool CloudClass::publish(const char*, PublishFlag)
     inline bool publish(const char *eventName, PublishFlag eventType=PUBLIC)
                 ^
../wiring/inc/spark_wiring_cloud.h:222:17: note:   candidate expects 2 arguments, 5 provided
../wiring/inc/spark_wiring_cloud.h:227:17: note: bool CloudClass::publish(const char*, const char*, PublishFlag)
     inline bool publish(const char *eventName, const char *eventData, PublishFlag eventType=PUBLIC)
                 ^
../wiring/inc/spark_wiring_cloud.h:227:17: note:   candidate expects 3 arguments, 5 provided
../wiring/inc/spark_wiring_cloud.h:232:17: note: bool CloudClass::publish(const char*, const char*, PublishFlag, PublishFlag)
     inline bool publish(const char *eventName, const char *eventData, PublishFlag f1, PublishFlag f2)
                 ^
../wiring/inc/spark_wiring_cloud.h:232:17: note:   candidate expects 4 arguments, 5 provided
../wiring/inc/spark_wiring_cloud.h:238:17: note: bool CloudClass::publish(const char*, const char*, int, PublishFlag)
     inline bool publish(const char *eventName, const char *eventData, int ttl, PublishFlag eventType=PUBLIC)
                 ^
../wiring/inc/spark_wiring_cloud.h:238:17: note:   candidate expects 4 arguments, 5 provided
make[1]: *** [../build/target/user/platform-6test.o] Error 1
make: *** [user] Error 2

```

Excepted: compiles fine
Actual: compilation failed

m-mcgowan, 2016-12-13:
The first two arguments are strings - the event name and the event data, so the first `10` should be `"10"`

kennethlimcp, 2016-12-13:
copy-pasta error. Updated the test case (that still doesn't compile)

m-mcgowan, 2016-12-13:
Good catch Kenneth...looks like the 5 arg version is missing. This is a quick fix, so we'll get it out in the next release.  Although please keep in mind that I don't believe TTL is actually honoured at present, so the workaround is to simply remove it.

kennethlimcp, 2016-12-13:
Definitely, but the docs said otherwise. 🗡

smpickett, 2016-12-23:
There still appears to be a problem with the documentation that needs to be updated. Specifically:

int temperature = sensor.readTemperature();  // by way of example, not part of the API
Particle.publish("t", temperature, NO_ACK);
Particle.publish("t", temperature, PRIVATE, NO_ACK);
Particle.publish("t", temperature, ttl, PRIVATE, NO_ACK);

should be changed so that the 'temperature' argument is a string. No?

technobly, 2016-12-23:
Yes!  Thanks @smpickett, will do it now.

technobly, 2016-12-23:
And it's live: https://docs.particle.io/reference/firmware/electron/#particle-publish- and https://prerelease-docs.particle.io/reference/firmware/electron/#particle-publish-

closed 2017-03-14 by technobly

JohnKozell, 2017-09-30:
I'm compiling to electron and 0.6.2, yet I'm still getting the following error:
`no matching function for call to 'CloudClass::publish(const char [8], char [255], int, const PublishFlag&, const PublishFlag&)`

Am I missing something?

My call is:
`  Particle.publish("reading", payload, 60, PRIVATE, NO_ACK);`
Where payload is defined as:
`char payload[255];`

sergeuz, 2017-10-06:
@JohnKozell The fix is available starting with the 0.7.0-rc.1 release

PR #1236 (2017-03-14): Particle.publish() flags fixes
This PR sanitizes `Particle.publish()` overloads, making it possible to specify one or two flags for any variant of this method. The PR also implements a generic type-safe wrapper for enum-based flags, aka `particle::Flags<T>`, which enables regular syntax for ORed flag combinations as an option:

```cpp
Particle.publish("event", "data", PRIVATE | NO_ACK);
```

Fixes #1201 and #1194.

---

Doneness:

- [X] Contributor has signed CLA
- [X] Problem and Solution clearly stated
- [x] Code peer reviewed
- [x] API tests compiled
- [ ] Run unit/integration/application t [... cut]