particle-iot/device-os #1219: Allow setting/overriding DNS on Electron & Photon

vshymanskyy, 2016-12-26:
The firmware API is missing functionality for setting/overriding DNS on Electron & Photon.
Default DNS causes troubles especially on Electron, as for some reason the IP addresses are cached and not refreshed for looong time.

technobly, 2017-02-06:
Thanks for the request!  We'll look into the feasibility of this.  I'd like to understand your issue on the Electron better though.  What is your use case that requires a faster/more-updated DNS, and is this a public or private DNS that you'd like to switch to?  It would be interesting to run a test to compare the speed differences that you are seeing.  Please provide more information so that we can test.

m-mcgowan, 2017-10-22:
The link above doesn't look correct - it links to the `electron` project for building cross-platform GUIs - same name but a different project and purpose!

FuzzOli87, 2018-06-02:
@technobly 

I know this is an old issue but I'm wondering if anything has changed since then. What we are seeing is a TCP server sitting behind an AWS ELB. AWS assigns dynamic IPs to it's DNS names. What we are seeing is that when those IPs change, DNS resolution does not attempt to obtain the new IPs and seemingly relies on some cache.

WildXBird, 2021-04-06:
It would be better to have a method to intercept DNS requests and return the IP specified by myself.