particle-iot/device-os #1233: 0.6.0/0.6.1-pc.1 System.sleep(30) never wakes

marc-gist, 2017-01-19:
Sleep is broken again! 

```
void setup() {
    delay(8000);
    System.sleep(30);
}
void loop() {

}
```

expected result: device wakes and reconnects after 30 seconds
observed result: device keeps radio off (breathing white) indefinitely

avtolstoy, 2017-01-19:
Photon/P1/Core/Electron?

There was a fix for this in 0.6.1-rc.1: https://github.com/spark/firmware/pull/1156

marc-gist, 2017-01-19:
Sorry; Photon, i've not tried on the Electron.

avtolstoy, 2017-01-19:
I've just tested this on Photon with 0.6.1-rc.1 and it works as intended. It is a confirmed bug on 0.6.0 though. Please make sure that your Photon is at 0.6.1-rc.1 (e.g. by using CLI: `particle serial inspect`)

marc-gist, 2017-01-19:
I'll double check the firmware and re-test with the simple case I posted.

technobly, 2017-01-19:
I just tested this as well and can confirm my Photon was not waking up after 30 seconds, or 10 seconds.  It could be an intermittent issue because the [tests did pass for 0.6.1-rc.1 on my photon](https://github.com/spark/firmware/blob/develop/user/tests/wiring/sleep/sleep.cpp) and it was back online at the end.

ScruffR, 2017-01-21:
I'd even propose to hide that "pseudo sleep" from the docs and rather advise people to take control over the radio modules in their own hands - I never really saw the use of that "feature" anyway.

avtolstoy, 2017-01-22:
I've just ran a test for ~4 hours with 0.6.1-rc.1 and couldn't reproduce it. @technobly, could you confirm that it fails for you on **0.6.1-rc.1**?

marc-gist, 2017-01-22:
Yes... Fails on both my Photons running 0.6.1-rc.1
Updated from website.
Anyway I can get a debug log on serial output?

technobly, 2017-01-23:
Confirmed failing in Web IDE 0.6.1-rc.1 and locally compiled with `release/v0.6.1-rc.1` and GCC v4.9.3

sergeuz, 2017-01-27:
I left the test below running for an hour and it didn't fail (v0.6.1-rc.1, gcc 4.9.3). Looks like the issue depends on network environment/settings?

```cpp
#include "application.h"

namespace {

Serial1LogHandler logHandler(115200, LOG_LEVEL_WARN, {
    { "app", LOG_LEVEL_ALL }
});

void onConnectionStateChanged(system_event_t event, int data) {
    switch (event) {
    case cloud_status: {
        switch (data) {
        case cloud_status_disconnected:
            Log.info("cloud_status: disconnected");
            break;
        case cloud_status_connecting:
            Log.info("cloud_status: connecting");
            break;
        case cloud_status_connected:
            Log.info("cloud_status: connected");
            break;
        case cloud_status_disconnecting:
            Log.info("cloud_status: disconnecting");
            break;
        default:
            break;
        }
        break;
    }
    default:
        break;
    }
}

enum State {
    DELAY,
    SLEEP,
    CHECK,
    FAILED
};

State state = CHECK;
uint32_t t = 0;
int n = 0;

} // namespace

STARTUP(System.on(cloud_status, onConnectionStateChanged))

void setup() {
    t = millis();
}

void loop() {
    if (state == FAILED || millis() < t) {
        return;
    }
    t = millis();
    switch (state) {
    case DELAY: {
        Log.info("%d ====================", ++n);
        const int msec = random(1000, 10001);
        Log.info("delay %dms", msec);
        state = SLEEP; // next state
        t += msec;
        break;
    }
    case SLEEP: {
        const int sec = random(1, 61);
        Log.info("sleep %ds", sec);
        System.sleep(sec);
        state = CHECK;
        t += sec * 1000 + 8000; // 8 seconds to connect to the cloud
        break;
    }
    case CHECK: {
        if (Particle.connected()) {
            Log.info("connected");
            state = DELAY;
        } else {
            Log.error("FAILED");
            state = FAILED;
        }
        break;
    }
    default:
        break;
    }
}
```

technobly, 2017-01-28:
@sergey wow, nice test!  Did you also try running the two line example in the first post? Also our own wiring/sleep test doesn't fail for me, but the two line test in setup or loop does.

sergeuz, 2017-01-28:
@technobly yeap, original example and few variations of it worked for me as well

technobly, 2017-01-28:
I'll try your test and see if that fails for me.

technobly, 2017-10-24:
I just tried the original two line example above on 0.7.0-rc.3 and it works as expected.  As noted above our built in `wiring/sleep` wasn't previously failing for me while the simple two line code posted above was.  I'll have to go back to 0.6.0/0.6.1-rc.1 and see if this still fails in my environment to double check, but we may have a fix in place somewhere in between 0.6.1 and 0.7.0-rc.3.

technobly, 2017-10-24:
Going back to 0.6.0 vs 0.6.1-rc.1 (and 0.6.1), the two lines of code fails on 0.6.0, but works as expected on 0.6.1-rc.1 and 0.6.1.  Therefor I can't run the test @sergeuz posted on a failing system (0.6.0) as cloud_status is not supported until 0.6.1-rc.1.  

TL;DR: This issue appears to be resolved, but also does appear to be environment related since I previously tested this as failing on 0.6.1-rc.1.

technobly, 2017-10-24:
@marc-gist would you please close the issue if it is resolved for you on 0.6.1 or 0.7.0-rc.3?

closed 2020-03-03 by avtolstoy