particle-iot/device-os #1238: Arduino compatibility removed word type from main firmware

monkbroc, 2017-01-25:
`word` is no longer a keyword in 0.6.0. This is a regression for apps that use this keyword.

From http://community.particle.io/t/particle-libraries-v2-0-in-particle-cli-1-19-1-drawbacks/28839:
> The compiler considers the 32-bit data type "word" as unknown now, maybe due to compatibility with 8-bit Arduino platform. However, this data type is still present as a legal data type in the firmware reference documentation.

There was also a comment by @ScruffR about `word` in the latest firmware
https://github.com/spark/firmware/pull/1191#issuecomment-264219189

Test case to reproduce:

Copy-paste in build
```
word foo;
void setup() {}
```

See the error `error: 'word' does not name a type`

---

Completeness:
- [X] Minimum test case added
- [X] Device, system and user firmware versions stated
- [X] Particle confirmed

technobly, 2017-01-25:
This is fixed in 0.6.1-rc.2 which is out on build.particle.io.  `word` would also not be added by default in 0.6.1-rc.1, and would require `#include Arduino.h`.

monkbroc, 2017-01-25:
Can we add a compile test for `byte`and `word` to prevent a future regression?

technobly, 2017-01-25:
We do need to add a BUNCH of API tests for Arduino compatibility that was introduced in 0.6.1-rc.1, and 0.6.1-rc.2.

I don't think `word` compatibility regressed on 0.6.0 though, did it?  It doesn't compile in 0.5.3 or 0.5.0... when did it work before 0.6.1-rc.x?

monkbroc, 2017-01-25:
Hehe, yeah.

You are a right about `word` before 0.6.0. It's not a regression. It never worked before.

technobly, 2017-01-25:
Can we close this with the assumption it's fixed in 0.6.1-rc.x, and will be out in 0.6.1 default soon?

closed 2017-01-25 by monkbroc