particle-iot/device-os #1239: Hang in Build when selecting Library on Safari Mac.

kaspesla, 2017-01-25:
I can add other libraries we tried, but when I click on ADAFRUIT_ST7735, I only get "Loading..." forever.
It works okay in Chrome. The issue seems to be only with Safari.

1) In Safari on Mac, create a new App in build
2) Click on Libraries
3) Search for ADAFRUIT_ST7735
4) Click on it.
5) Observe not-workingness

Mac OS X 10.11.6
Safari Version 9.1.3 
---

Completeness:
- [ ] Minimum test case added
- [ ] Device, system and user firmware versions stated
- [ ] Particle confirmed

monkbroc, 2017-01-25:
Hang tight, we have a fix ready to deploy for this!

kaspesla, 2017-01-25:
Good to hear! Thanks.
Not a show-stopper since it does work in Chrome.
-Ken

monkbroc, 2017-01-25:
This is fixed now.

closed 2017-01-25 by monkbroc