particle-iot/device-os #1240: TCP Firmware sends ACKs and separate response for every Chunk in Fast OTA mode

brycekahle, 2017-01-25:
In Fast OTA mode, the firmware should not respond to every Chunk message (especially because they are NON confirmable messages). To compensate for this, it should respond to the UpdateDone with whether it needs to request additional missed/invalid chunks. Those missing/invalid chunks are then requested via the ChunkMissed message.

The current firmware follows the correct procedure for requesting missing chunks in Fast OTA mode, but it **incorrectly sends ACKs and separate responses for each individual Chunk message**.

Firmware version: up to and including 0.6.1-rc.2
Device: Photon/P1/Core (any TCP device)

---

Completeness:
- [ ] Minimum test case added
- [ ] Device, system and user firmware versions stated
- [ ] Particle confirmed

towynlin, 2017-01-25:
Actually I think the file `spark_protocol.cpp` is only used for the Core. I'm pretty sure the Photon, P1, and any other TCP device uses the `lightssl_message_channel`, `lightssl_protocol`, and then just like the DTLS firmware path, uses [ChunkedTransfer](https://github.com/spark/firmware/blob/dfdadfb3e3509bf3e98a28f64bd3917caca1ffa2/communication/src/chunked_transfer.cpp#L139) to manage OTA.

@m-mcgowan can you confirm/deny?

avtolstoy, 2017-01-25:
@towynlin TCP devices still use `SparkProtocol`. See https://github.com/spark/firmware/blob/develop/communication/src/protocol_selector.h#L24

towynlin, 2017-01-25:
Ah thanks Andrey. So... does nothing use lightssl*?

avtolstoy, 2017-01-25:
Currently no :)

towynlin, 2017-01-25:
LOL, OK, well then the path forward for working on this issue is clear! 😺

PR #1270 (2017-05-09): [ch1741] OTA'd module info sent as a payload in UpdateDone message or as an ACK to UpdateDone
Original PR #1097 for issue #1032 introduced `spark/device/ota_result` event that was posted after receiving a binary OTA and before performing a reboot, which carried OTA'd module info after validity has been checked. This PR removes that separate event and instead sends OTA'd module info as a payload in UpdateDone message or as an ACK to UpdateDone as disccussed in _ch1741_.

This PR also fixes #1240.

TODO: Update acceptance test (#1153) if possible.

---

Doneness:

- [X] Contributor has signed CL [... cut]
commit 64e32e25 (2017-05-09): Removes spark/device/ota_result event. OTA'd module description is sent as UpdateDone message or as a piggy-back ACK to UpdateDone.
Also fixes #1240
closed 2017-05-09 by technobly