particle-iot/device-os #1241: WiFi.connecting() returns `false` during DHCP resolve

technobly, 2017-01-25:
### Given:

When a Wi-Fi device connects to a Wi-Fi AP, the system RGB LED will blink slow green while it connects to the router with it's credentials, and then fast green while DHCP is resolving.  After DHCP resolves the device has an IP address and `WiFi.ready()` will return `true`.

### Expected Behavior:

`WiFi.connecting()` returns `true` while Wi-Fi connection process is underway, and `false` after a successful connection is made.

### Problem:

`WiFi.connecting()` returns `false` while DHCP is resolving.

### Example:

```c++
SYSTEM_MODE(MANUAL);
SYSTEM_THREAD(ENABLED);

void setup() {
    pinMode(D7, OUTPUT);
    Particle.connect();
}

void loop() {
    // During a SLOW DHCP resolve, D7 will turn off briefly while the RGB is flashing fast green
    if (WiFi.connecting() || WiFi.ready()) {
        digitalWrite(D7, HIGH);
    }
    else {
        digitalWrite(D7, LOW);
    }
    Particle.process();
}
```

Thanks @suda !

---

Completeness:
- [x] Minimum test case added
- [x] Device, system and user firmware versions stated (0.6.1-rc.2)
- [x] Particle confirmed

sergeuz, 2017-02-05:
@m-mcgowan What would be better, to make `WLAN_CONNECTING = 1` span over both AP connection and DHCP phases, or split `WLAN_DHCP` into two flags, e.g. `WLAN_DHCP_PENDING` and `WLAN_DHCP_DONE`?

m-mcgowan, 2017-02-05:
I feel `WLAN_CONNECTING` should span the entire connection process since that's what is least surprising.

closed 2017-03-21 by technobly

suda, 2017-03-21:
[image]

PR #1254 (2017-03-21): Fix for WiFi.connecting() returning false during DHCP configuration
This PR fixes #1241 by making `WLAN_CONNECTING` and `WLAN_CONNECTED` flags span over the entire network connection process, including DHCP configuration phase.

I'm not sure how I can write a good test for this fix, so below is my usual test application which I play with to debug network/cloud connectivity problems. Make sure you build the system firmware with debugging enabled and `DEBUG_NETWORK_STATE` macro defined to get internal state of the `ManagedNetworkInterface` class logged.

Tested on Photon and Electron with static [... cut]