particle-iot/device-os #1243: UDP.status() to provide socket status

barakwei, 2017-01-28:
`TCPClient::status` returns the status of the socket, while `UDP` doesn't have such method.
---

Completeness:
- [x] Minimum test case added
- [x] Device, system and user firmware versions stated
- [ ] Particle confirmed

technobly, 2017-02-06:
Thanks for the request!

barakwei, 2017-02-08:
Is there a workaround I can use for now?

technobly, 2017-02-08:
I think the best you can do right now is double check that Wi-Fi is ready `WiFi.ready()`.  I'm not sure what errors we will be able to detect for UDP since it's a connectionless protocol.  There are errors that can be generated, but do we see them in all cases and which ones we choose to use to change UDP::status will be the question.

barakwei, 2017-02-09:
Using `WiFi.ready()` won't help. The socket dies after power transition, and WiFi is fine.
I Guess the best thing I can do right now is if `sendPacket()` fails, is to call `UDP.begin()` again.

I expect UDP::status to be implemented like TCP::status. Basically is the socket alive.

justinatomatic, 2017-02-09:
The socket also dies if the photon decides to reset the network stack while reconnecting to the cloud.  Having a UDP::status would be great.