particle-iot/device-os #1244: Cellular.RSSI() implementation/documentation of qual is not correct

asallam, 2017-01-29:
for UMTS RAT the Cellular.RSS() function should return rssi in dBm which is ok.
However, the (Energy per chip/Noise) as explained in the documentation cannot be correct as described in dB. 
qual: (int) is a number in UMTS RAT indicating the Energy per Chip/Noise ratio in dB levels of the current cell. This value ranges from 0 to 49, higher numbers indicate higher signal quality.

the proper range of reporting as in 3GPP standards is between 0dB to -24dB in 0.5dB steps which result in 50 different levels.
49 is the best ECNO corresponding to 0dB
0 is lowest signal corresponding to <-24dB.
on the other hand, the 3G modem of u-blox-270 maps this range to 8 levels only [1]

Reference: 
[1] u-blox Cellular Modules, Data and Voice Modules, AT Commands Manual
Document number UBX-13002752
Revision and date R47 23-Dec-2016

---

Completeness:
- [ ] Minimum test case added
- [ ] Device, system and user firmware versions stated
- [ ] Particle confirmed

technobly, 2017-01-29:
Thanks for the report! At the time this was implemented the AT command manual didn't have as much information as it does now. So at a minimum we can clarify that the qual value corresponds to a different range in dB values. It gets a little more complicated though because those values and the meaning of qual can change based on the current RAT you are using. This can be at least 3 different types on the U260 and U270. It might not be so important to pin these exact values down and just know that a higher qual number equals higher signal quality, and the range is 0-49. I think we can fix this in documentation.

avtolstoy, 2020-03-03:
https://docs.particle.io/reference/device-os/firmware/electron/#cellularsignal-class

closed 2020-03-03 by avtolstoy