particle-iot/device-os #1249: Weird status LED freezing (Photon, 060)

jenesaisdiq, 2017-02-12:
I'm running a blocking loop with threading enabled and sometimes the status LED will stop updating for a long period- tens of seconds. Code uses the InternetButton lib. Not going to take the time right now to strip it down to minimum reproduceable.

```
SYSTEM_THREAD(ENABLED);

STARTUP(WiFi.selectAntenna(ANT_EXTERNAL));
#include <InternetButton.h>

InternetButton b;

void setup()
{
  b.begin();
  b.setNumLeds(27);
  pinMode(D4,INPUT_PULLUP);
  pinMode(D7,INPUT_PULLUP);  
}

void loop()
{
    float i = 28;
    for(float j = 11; j<19; j=j+.01){
        b.smoothLedOn(j,0,20,255);
        b.smoothLedOn(i,0,20,255);
        delay(1);
        b.smoothLedOn(j,0,0,0);
        b.smoothLedOn(i,0,0,0);
        i-=.01;
    }
}
```

---

Completeness:
- [x ] Minimum test case added
- [x ] Device, system and user firmware versions stated
- [ ] Particle confirmed

jenesaisdiq, 2017-02-12:
Appears to cause interruptions/blocking while the device is being flashed. The LED will be magenta or off for longer periods than expected, and I've also seen flashes appear to complete (many, many cycles of magenta/off) and then reconnect but WITHOUT running the new code... so may cause failed flashes.

technobly, 2017-02-12:
Please try unrolling your for loop and not using hard delays.  Also every time the neopixel library is called to update the strip it will disable interrupts to give strict focus to GPIO timing which will cause the system LED to stop updating. By unrolling the for loop, it should significantly increase the update rate of the system LED. Use millis() as well for a 1ms update rate instead of the hard delay.

jenesaisdiq, 2017-02-13:
Oh sure, I *know* it's garbage code, was written hackily and hastily to test an LED pattern. The bug bit is that I'd hope the status LED wouldn't act up with blocking code while in threaded mode, which gave me concerns for "multithreaded by default". You think it's more likely to be a problem with neopixel disabling interrupts?

technobly, 2017-02-13:
Yeah, it would be easy to test the difference by either creating the same hard delay and then doing it with interrupts disabled or not.

```c
SYSTEM_THREAD(ENABLED);

void setup()
{ 
}

void loop()
{
    float i = 28;
    for(float j = 11; j<19; j=j+.01){
        // test disabling interrupts as neopixel does, 
        // or keep them enabled them by commenting out
        __disable_irq();
        // simulate neopixel delay
        delay(1);
        // now re-enable interrupts
        __enable_irq();
        // same delay from before
        delay(1);
        i-=.01;
    }
}
```

@jenesaisdiq I just tested this and it completely blocks system operation 100% if interrupts are only-disabled, however as written above it seems ok even though it is now blocking the system 50% of the time.  I wonder if we should generate a warning when disabling interrupt code is detected, and make sure users are aware that it will negatively impact parts of system operation.  Users can disable interrupts if they want to, and will need to in certain circumstances... so we'll have to educate about this potential issue.  I'm not sure why you saw as much blocking as you did though.

closed 2020-03-03 by avtolstoy