particle-iot/device-os #1251: Change default DHCP client name

brycekahle, 2017-02-13:
The default is `WICED DHCP Client` which can cause problems with DHCP servers who also automatically add a DNS hostname because it contains spaces.

---

Completeness:
- [ ] Minimum test case added
- [ ] Device, system and user firmware versions stated
- [ ] Particle confirmed

technobly, 2017-02-22:
@brycekahle Is the recommendation here just to remove the spaces from the name?  Also do you have a test case where we can verify this change would be successful?

brycekahle, 2017-02-22:
Definitely remove the spaces. It might be nice to use something that makes it clear what device it is, say `ParticlePhoton`?

avtolstoy, 2017-04-06:
What if we use device id as a default hostname? Sounds like a good option as well, considering device id is unique.

We should probably also add some setter/getter methods to modify hostname (e.g. `WiFi.get/setHostname()` or `System.get/setHostname()`)

avtolstoy, 2017-07-07:
Implemented in #1296. Release in 0.7.0-rc.1.

closed 2017-07-07 by avtolstoy