particle-iot/device-os #1252: first TCP connection to softap is not closed automatically by the device.

m-mcgowan, 2017-02-15:
The device should close it's end of the TCP connection after sending the response data. This doesn't happen for the first connection, but does happen on subsequent connections. 

Expected outcome
```
Given I have photon in listening mode
And the computer's Wi-Fi is connected to the Photon's AP
When I run `echo -e -n "device-id\n0\n\n" | nc 192.168.0.1 5609`
Then the device should output some JSON
And the stream is closed
And When I run `echo -e -n "device-id\n0\n\n" | nc 192.168.0.1 5609` again
Then the device should output some JSON 
And the stream is closed
```

Actual outcome

```
Given I have photon in listening mode
And the computer's Wi-Fi is connected to the Photon's AP
When I run `echo -e -n "device-id\n0\n\n" | nc 192.168.0.1 5609`
Then the device should output some JSON
And the stream remains open
When I press Ctrl-C the command exits
And When I run `echo -e -n "device-id\n0\n\n" | nc 192.168.0.1 5609` again
Then the device should output some JSON
And the stream is closed
```

What happens in practice is the stream from the first command is not closed. Instead, you have to use Ctrl-C to close the stream.  Subsequent invocations of the same command then sees the device correctly close the stream automatically. 

Tested on a Photon, firmware 0.4.9 and 0.6.0
---

Completeness:
- [X] Minimum test case added
- [X] Device, system and user firmware versions stated
- [ ] Particle confirmed

sergeuz, 2017-03-03:
There's a bug in WICED, which erroneously treats a socket with zero internal index as an invalid one: https://github.com/spark/photon-wiced/blob/master/WICED/network/LwIP/WICED/tcpip.c#L224.

I also encountered another issue while trying your netcat example: WICED may generate an additional (and invalid) connection event, causing `TCPServerDispatcher` to hang trying to accept an invalid connection – right after processing a TCP request successfully.

Both issues seem to remain unfixed in WICED 3.7.0.

avtolstoy, 2017-07-07:
Fixed in spark/photon-wiced#6. Possibly a related softap issue fixed by 596f73891708d167fa76bb70a04b331c29ac5678. Released in 0.7.0-rc.1

closed 2017-07-07 by avtolstoy