particle-iot/device-os #1253: Define platform constant names for Bluz, Oak, RedBear and Raspberry Pi

monkbroc, 2017-02-15:
We want people to start using the platform definition constants in their libraries instead of the numerical platform ID but several platforms don't have a constant defined in https://github.com/spark/firmware/blob/develop/platform/shared/inc/platforms.h

Add Bluz, Oak, RedBear and Raspberry Pi to platforms.h

---

Completeness:
- [ ] Minimum test case added
- [ ] Device, system and user firmware versions stated
- [ ] Particle confirmed

m-mcgowan, 2017-02-15:
This interesting. I appreciate this is a quick fix! 

But I'm not sure this should be in firmware, or we need to do a firmware release to support new compounds.  I'm thinking this could be stored in a separate repo from the firmware repo, so that we can add new platforms without having to do a firmware release.

We could create a compounds repo that has definitions for all the platform constants, and also feature flags for each compound. And then I'm thinking well...isn't that part of what's in the compound's own repo?

Not sure what the solution is...just thinking aloud and hoping to seed an interesting discussion!

monkbroc, 2017-02-15:
I see your point. Libraries couldn't use new platform definitions and remain compilable on older firmware.

Here's Brett's docs issue https://github.com/spark/docs/issues/413 for reference.

m-mcgowan, 2017-02-15:
Yeah, the supported compound platforms are related to firmware revisions, but on a different axis of development to firmware. So it's entirely possible that a compound is created for older versions of firmware, not just the latest release.

technobly, 2017-02-23:
PLATFORM_ID can always just be compared to a numerical value as well.  In library guidance docs we can document the compound platform id's:
```
Bluz = 104
RedBear Duo = 88
RasPi = 31
Oak = 82
```

monkbroc, 2017-02-23:
It has always felt wrong asking people to use magic numbers in their
libraries instead of symbols.I was looking for a solution.

*Julien Vanier*
 | julien@particle.io | cell 248-481-0771 | www.particle.io

technobly, 2017-02-28:
To summarize and keep the discussion going:

- ✅  Add platform ID/Compound lookup to Docs.  👎  _Requires users to use ID's._  This doesn't seem like a bigger barrier over having to know the platform name though.  We should do this anyway.
- ❓ Add compound platform names to system firmware.  👎  _This won't help compounds/library creators if they are not keeping up to date with system firmware._  Compounds could back-patch their own maintained firmware though.
- ❓ 🆕  Add compound platform names as exports to Cloud environment so that they are available for all firmware versions.  No need to maintain in system firmware, can update at any time for all firmware.  👎  _only available for Cloud compilation._  If adding dependencies to the firmware repo wasn't so bad, we could add this information to the firmware repo as a subtree/submodule, but that seems like a lot of maintenance overhead for some simple defines.  Maybe it could be easily added to Buildpacks though, so there would be a solution for the local toolchain.

closed 2020-03-03 by avtolstoy