particle-iot/device-os #1257: softap-setup equivalents over USB serial (and custom setup over other conduits)

m-mcgowan, 2017-02-20:
In order to support complete setup of a device via USB, we need the soft AP functionality (available via TCP/HTTP) to be available via USB.

The softAP functionality is

- scanning for wifi networks (not supported over USB)
- determining and setting the active antenna (not supported over USB)
- retrieving the device ID (supported over USB)
- setting the claim code and determining if a device is claimed (already supported over USB)
- setting wifi credentials (already supported over USB in plaintext)
- retrieve the device's public key (not supported)
- naming the device (currently not supported via any interface.)

Since we assume USB is a secure connection, there is no need to encrypt the wifi password with the device's public key. However, we should still provide the public key via USB in case it is required in future. 

Naming the device is a afterthought - the setup process could provide the device name, and the device name is sent to the cloud when the device is claimed. This allows the device details to be given when the device is offline, to be applied later when the device is online. 

Completeness:
- [ ] Minimum test case added
- [ ] Device, system and user firmware versions stated
- [ ] Particle confirmed

mebrunet, 2017-02-20:
Two thumbs up. This would be great.

As for claiming over USB, is there anything we should know to make it work right now? I'm having [issues](https://github.com/spark/firmware/issues/602#issuecomment-281183770).

avtolstoy, 2020-03-03:
https://github.com/particle-iot/firmware-protobuf

closed 2020-03-03 by avtolstoy