particle-iot/device-os #1261: Electron Sleep mode with time not waking when WKP left high

TrentonH1, 2017-02-27:
I'm using System.sleep(SLEEP_MODE_DEEP,15). It's my understanding that the device will wake up after 15 sec or after a rising signal to the WKP pin. It will wake up under both of those conditions, but if I wake up with the WKP pin and leave it high it will not wake up. It will only wake up with the reset button. I tried this with 10 and 3 seconds also with the same result. Is this a bug or is there something I don't know?

Rick from Particle also did these tests.

I did some experiments with this, and the behavior is a little weird.
•	Good: WKP is low when you go to sleep and you wake up by time
•	Good: WKP is low when you go to sleep and you wake up by bringing WKP high
•	Good: If WKP is high when you go to sleep, and you bring it low after sleep, and then wake up by time
•	Good: If WKP is high when you go to sleep, and you bring it low after sleep, and then bring it high again
•	Bad: If WKP is high when you go to sleep, and you leave it high past where you would have woken by time, it won't wake up
•	Bad: If WKP is high when you go to sleep, and you leave it high past where you would have woken by time, it won't wake up if you bring it low, then high again.
This is the code I used on a Photon running 0.6.0.

>`#include "Particle.h"
>
>void setup() {
>    Serial.begin(9600);
>}
>
>void loop() {
>    delay(5000);
>    System.sleep(SLEEP_MODE_DEEP, 15);
>}
---

Completeness:
- [x ] Minimum test case added
- [x ] Device, system and user firmware versions stated
- [x ] Particle confirmed

technobly, 2017-02-28:
I accidentally duplicated this issue #1262 while confirming a post in the community.

m-mcgowan, 2017-10-03:
Thank you for the report @TrentonH1.  Since we have a duplicate that has some comments, I'll closing this and we'll track progress in #1262.

closed 2017-10-03 by m-mcgowan