particle-iot/device-os #1262: If WKP pin is held HIGH on deep sleep wake, RTC alarm and future LOW to HIGH transition on WKP will not wake device.

technobly, 2017-02-28:
If WKP pin is held HIGH on deep sleep wake, RTC alarm and future LOW to HIGH transition on WKP will not wake device.

Tested on 0.6.0 and 0.6.1-rc.2.

- If you short WKP to 3V3 and let the device sleep, you can observe:
  - remove short before 5 seconds expires without accidentally glitching the WKP LOW/HIGH and the RTC alarm will wake the device.
  - remove short before 5 seconds expires without accidentally glitching the WKP LOW/HIGH and short WKP back to 3V3 the device will wake.
  - leave short for > 5 seconds and RTC alarm will fail to wake device, also removing the short and reapplying (LOW to HIGH transition on WKP) will not wake the device.
  - Only affected when RTC is used, does not occur with `System.sleep(SLEEP_MODE_DEEP);`

See community post here: https://community.particle.io/t/alarm-issue-with-sleep-mode-deep-after-wkp-pin-event/30018
>There is an interesting Note in the STM32F205xx Reference Manual which states the following:

>Note: An additional wakeup event is detected if the WKUP pin is enabled (by setting the EWUP bit) when the WKUP pin level is already high. (see PWR_CSR register description).

```c
void setup()
{
    pinMode(D7, OUTPUT);
    digitalWrite(D7, HIGH);
    delay(1000);
    System.sleep(SLEEP_MODE_DEEP, 5);
}

void loop()
{
}
```

---

Completeness:
- [x] Minimum test case added
- [x] Device, system and user firmware versions stated (as of 0.6.0 and 0.6.1-rc.2)

technobly, 2017-02-28:
Duplicated here: https://github.com/spark/firmware/issues/1261

sergeuz, 2017-03-02:
It looks like we may need help from ST to resolve this issue, or confirm that this is a hardware bug. I've double-checked the sequence of operations we're doing to enter standby mode with the reference manual, tried various workarounds (thanks, @avtolstoy) and, as a last resort, simply copied ST's example code for standby mode (it uses the same RTC alarm and enables WKUP pin) into our firmware to ensure that we configure RTC in similar way, clear flags in the same order, etc. – the issue still persists :(

Toklandarr, 2017-08-01:
@sergeuz is there an overview of possible Workarounds? I may have the same issue with a rather involved setup.
Could you for example just reset using Particle.reset() when you wake up and WKP is high?
Thanks in advance!

sergeuz, 2017-08-02:
@Toklandarr I have no further info unfortunately. Last time I checked, the issue persisted regardless of any workarounds I tried. ST's errata for the chip didn't help either.

jschwalbe, 2018-02-26:
I'm trying a workaround to check if it's moving before going to sleep. I'd like to make it wait 5 minutes and then check again, etc. But I can't find a reliable way to check for "is there motion RIGHT NOW". Any ideas?

pkourany, 2018-08-08:
The Jan 2015 errata from ST Micro states:

[image]

Doing the work-around leads to the immediate reset of the device if the (one of) the wake up sources (WKP pin) is HIGH when entering deep sleep.  Bottom line seems to indicate that the WKP pin _cannot_ be HIGH when entering deep sleep mode.

ScruffR, 2018-09-23:
Any update on the issue? 
This renders devices like Particle's own AssetTracker shield with WakeOnMove virtually useless since you can't guarantee that there migh not be any transient HIGH signal at the very moment of engagint sleep mode.

I'm running a modified (deep sleep) version of the AssetTrackerRK 3_WakeOnMove example and eventually *always* get unresolvable permanent sleep phases.

If this bug can't be solved on the µC side, the AssetTracker Shield would need a redesign to allow for external HW side workaround (e.g. temporary disabling and "latching" accelerometer interrupts).

ericyuanyong, 2018-09-25:
I also want to know what's the fix solution, we are going to use this feature for customer's project. So a little bit urgent to know the solution

technobly, 2018-09-25:
Isn't the issue not that when you go to sleep the WKP pin is HIGH, but that **after you are asleep when the RTC timer expires, if the WKP is already HIGH that it won't wake the MCU**?

If so, one way to minimize the potential of WKP being HIGH on RTC timeout is to not allow WKP to be a DC signal.  So you would AC couple it to the WKP pin in such a way that it can still wake the STM when the pin goes from LOW to HIGH (surely there is a timing requirement here?), but immediately after it falls back LOW until whatever is driving it is reset back to LOW.

Another way to fix the problem would be to leverage an external HW WDT to have a similar but longer timeout to that of the deep sleep RTC value.  So if your HW WDT was 120 seconds, and your deep sleep command was 90 seconds... if this issue prevented you from waking up, the HW WDT would hard reset the STM32 and you'd be back in business.  This combined with the AC coupling of the WKP pin would mean you'd likely never see the HW WDT reset the part.

---

**If the issue is ALSO that the device fails to go to sleep if the WKP pin is held HIGH when deep sleep is issued** (something I don't believe I saw in my original testing above): 

>If this bug can't be solved on the µC side, the AssetTracker Shield would need a redesign to allow for external HW side workaround (e.g. temporary disabling and "latching" accelerometer interrupts).

Then I think this is probably the best guaranteed way of getting to sleep in the first place.

That said, if you are in motion you will just wake back up anyway based on the accelerometer.

One way to potentially combat this is to more heavily filter the accelerometer output, either with accelerometer configuration or by inserting a timer in-between the accelerometer output and the WKP input.  Let's say it's a low power 555 timer circuit (or equivalent) that will start timing after the accelerometer indicates motion for 10 seconds.  You should be able to then reset it via a GPIO just before going to sleep and have a guaranteed LOW input on WKP for the next 10 seconds.  Depending on how it gets reset, you might also be able to use it for anti-shock based nuisance tripping.  This kind of depends on how the accelerometer can be set to interrupt (which I think is currently latching based on a certain threshold crossed, which is relative to a calibrated offset).  What you really need is something that effectively debounces the still sensitive accelerometer reading.  You could possibly do this by waking up resetting the accelerometer's latched interrupt output and seeing if it fires again immediately, and if so stay awake this time.  With a system like that you probably wouldn't want the 555 timer "buffer" to delay too long.  Just long enough that you could reliably go to sleep in the first place because it's now part of the debounce interval.

sidwarkd, 2018-10-03:
+1 on experiencing this issue with a P1 module setup in my product. I use USB to wake my device by having the USB tied to the WKP pin through a resistor. If I enter deep sleep mode while still attached to USB my device never re-awakens.

cethalon, 2019-04-18:
Since there is no workaround I think it would be a good idea to make note of this interaction in the docs. It may save someone like me from learning about this interaction the hard way! 😉

closed 2020-03-03 by avtolstoy