particle-iot/device-os #1267: Move global defines to a namespace.

Lindenk, 2017-03-02:
Hello!

After the last release, `Promise` has now been declared globally. The project I'm currently working on has it's own `Promise` implementation which, of course, causes the entire thing to break.

My options are:

- Rename my own `Promise` implementation and every reference to it to something less descriptive
- Compile using an earlier release, which will not be sufficient long term
- Post an issue here to stop new releases from clobbering defines in existing projects

My feature request is to add everything defined globally into it's own namespace and have `.ino` files `using namespace spark` by default. As it stands now, to my knowledge, there is no way to override the globals defined in the firmware so, in cases like these where there is a new naming conflict, the end user is forced to modify and obfuscate a large portion of their code for it to build. By adding these globals to a namespace, the user can choose when and when not to use spark defines. 

Of course this may be a breaking change to many projects and libraries already existing, but without this, every new feature added to the firmware is potentially a breaking change.

I'm currently using:

Firmware version: `0.6.1`
Particle: `photon`

sergeuz, 2017-03-02:
I agree that we need to stop pulling everything into the global namespace unconditionally. There should be some macro like `DONT_USE_GLOBAL_NAMESPACE` that, once defined, will allow advanced users to get full control over visibility of particular Particle APIs in the global namespace, without breaking compatibility with existent applications.

monkbroc, 2017-03-07:
Here's another report: `Message` is a global symbol in Particle firmware defined [here](https://github.com/spark/firmware/blob/b9ce4a974dd9b900c45f2b3f5d8ba7cb0ebbd088/system/inc/active_object.h#L77). It's quite conceivable that user code might want to define a class called Message. The Particle firmware `Message` should be included in the `particle` namespace.

m-mcgowan, 2017-03-10:
I agree. I thought our C++ code was already namespaced.