particle-iot/device-os #1272: Expose LEDCloudSignalStatus class to user firmware

suda, 2017-03-09:
Firmware contains the `LEDCloudSignalStatus` class aka Nyan mode but it's not available in user firmware (all other system modes are). Instead of users copying and pasting this code it could be refactored to be a public class.

Example app:

```c++
LEDCloudSignalStatus rainbowSignal(LED_PRIORITY_IMPORTANT);

void setup() {
    rainbowSignal.setActive(true);
}

void loop() {
}
```

---

Completeness:
- [x] Minimum test case added
- [x] Particle confirmed

sergeuz, 2017-03-09:
Hey @suda, is there any practical reason why this class should be a part of the public API? Firmware reference has a section that describes how to implement similar custom patterns: https://docs.particle.io/reference/firmware/photon/#custom-patterns

suda, 2017-03-09:
It's already implemented there shouldn't be a need for copy/pasting this code. I don't see a reason why it shouldn't be public especially when other LED modes are available too :)

sergeuz, 2017-03-09:
My point is that the LED signaling API doesn't aim to provide a way for applications to mimic the system LED signaling in all its variety, it's a generic API that supports several built-in patterns like breathing and blinking colors (without relation to a network or any other system status), and allows users to implement their own custom patterns if necessary. `LEDCloudSignalStatus` is just one of such patterns that the system firmware implements for its own purposes along with `LEDCloudErrorStatus`, `LEDCounterStatus` and several other statuses.

suda, 2017-03-09:
I love this API! IMO "nyan" or the counting blinks are generic enough to be useful on their own. In my case I used nyan to indicate that the device is currently busy executing a task. I can see the counter status used for i.e. battery status too.

closed 2020-03-03 by avtolstoy