particle-iot/device-os #1274: String object gets re-initialized during setup() when using SYSTEM_THREAD(ENABLED);

alotaiba, 2017-03-09:
I have a display connected to Particle Photon. It's a custom LED matrix based on HT1635B LED driver. My issue is when I define a `String` object containing some text in `setup()`, I see garbled display. Sometimes I see part of the text, but it disappears quickly, as if it's being re-initialized.

I found three work arounds for this issue, which I explain below. But before that, here's the problematic code:

```Arduino
#include "LED_GFX.h"

LED_GFX gfx;

SYSTEM_THREAD(ENABLED);

void setup() {
  Serial.begin(115200);
  gfx.begin();

  String sBuffer = "Hello";
  gfx.display(sBuffer.c_str());
}

void loop() {
  gfx.run();
}
```

There are three solutions I discovered to solve the issue:
1. Define `sBuffer` globally before `setup()`.

        String sBuffer;

        SYSTEM_THREAD(ENABLED);

        void setup() {
          Serial.begin(115200);
          gfx.begin();

          sBuffer = "Hello";
          gfx.display(sBuffer.c_str());
        }

1. Remove the multithreading `SYSTEM_THREAD(ENABLED)` functionality.

        void setup() {
          Serial.begin(115200);
          gfx.begin();

          String sBuffer = "Hello";
          gfx.display(sBuffer.c_str());
        }

1. Start defining `sBuffer` and displaying the data during `loop()` (after some delay).

This has been found on 0.6.1. I'm not sure about earlier releases though.

---

Completeness:
- [ ] Minimum test case added
- [ ] Device, system and user firmware versions stated
- [ ] Particle confirmed

monkbroc, 2017-03-09:
This sounds like the LED_GFX library is holding on to the string longer than it should. Defining the string variable globally is the right way to do what you are trying to do. The fact that this currently crashes only with `SYSTEM_THREAD(ENABLED)` is chance.

m-mcgowan, 2017-03-09:
If you can share the library code you're using we might be able to verify the cause, but without that we can only guess. 

I too think you have a hidden memory leak here, that the gfx library is saving the pointer to the string (which is allocated on the heap), but the pointer memory is freed at the end of `setup()`.  It's only by chance that the memory is still valid by the time `loop()` is run in single-threaded mode, while multithreaded mode places more demands on the heap and so the corruption is more apparent.

Although turning off multithreading seems to avoid the issue, it's not working by design but by good luck...as the app becomes more complex and uses more of the heap, the chances of this failing increase. 

Moving the messages into global space fixes the problem since the memory is not reclaimed until the application closes.

However, for non-changing strings, the ideal solution is to not use "String" objects but simply use C strings for string constants, like this: `const char* message = "hello";`.  These strings are stored in flash memory which has several benefits:

1.  there is no allocation lifetime issue - the strings are available always
2. the strings do not take up any RAM since they are stored in flash. `String` allocates the storage in ram, plus the storage for the original C string in flash.

alotaiba, 2017-03-10:
I completely missed the point that `String` is actually allocated on the heap. I always thought that anything defined within a function will be allocated on the stack. Much thanks @monkbroc and Mathew for the clarification.

So just to clarify more about `String` class, if it's actually allocated on the heap, aren't users required to free it explicitly to prevent memory leaks, or does that happen automatically at the end of the function?

alotaiba, 2017-03-11:
Thank you @m-mcgowan, that cleared it for me.

closed 2017-03-11 by alotaiba