particle-iot/device-os #1276: USARTSerial::halduplex() seems to mix up TX data into RX buffer

ScruffR, 2017-03-11:
As discussed with @avtolstoy I use this code
```
const int srfAddress = 0x01;                      // default sensor address 

#define _USE_SOFT_SERIAL_  0

#if ( _USE_SOFT_SERIAL_ == 1 )
  #include "ParticleSoftSerial.h"  // needs 0.0.6 (not yet published)
  ParticleSoftSerial com(TX, TX);
#else
  USARTSerial& com = Serial1;
#endif

void setup()
{
  Serial.begin(115200);
#if ( _USE_SOFT_SERIAL_ == 1 )
  com.begin(9600);
#else
  com.begin(9600);
  com.halfduplex(true);
#endif
}

void loop()
{
  uint8_t  status = 0;
  uint16_t range  = 0;
  Serial.printlnf("Range  %d", srfSendCmd(srfAddress, 0x54, &range , 2));       // immediate read
  Serial.printlnf("Range: %d cm (%04x) (%s) - %02x", 
                  range, range,
                  ((range > 19) || (status & 0x01)) ? "reliable" : "uncertain", status);
                  
  delay(100);
}

int srfSendCmd(uint8_t Address, uint8_t cmd, void* buf, int len)
{                
  uint8_t* _buf = (uint8_t*)buf;
  uint8_t  _cmd[2] = { Address, cmd };
  
  //com.flush();
  while(com.read() >= 0);  // flush RX buffer
  com.breakTx();
  delayMicroseconds(1000);
  com.write(_cmd, 2);

  // wait for data
  for(uint32_t ms = millis(); (com.available() < len) && (millis() - ms < 100); Particle.process());

  // retrieve data
  if (com.available() >= len)
  { // big endian, needs to flip byte order (hence count down)
    while(len) _buf[--len] = com.read();
  }

  return -len;                                    // should be zero at the end
}
```
to communicate with an SRF01 range sensor, which works fine with my ParticleSoftSerial lib, but with Serial1 I keep getting the TX data as part of my RX response messing it up. 
I can't imagine this is the intended behaviour of half-duplex communication, or is it?

*(Photon 0.6.1)*
---

Completeness:
- [x] Minimum test case added
- [x] Device, system and user firmware versions stated
- [ ] Particle confirmed

technobly, 2017-04-07:
Thanks @ScruffR ! 

@avtolstoy have you confirmed this issue already?

avtolstoy, 2017-04-14:
@technobly @ScruffR I've looked into this a bit more. I don't think we should fix the default behavior with `halfduplex(true)`. Having echo enabled in half-duplex mode can be useful for instance when implementing 1-Wire using UART. A simple example is that If TX pin is correctly configured as open-drain and we are writing `0xFF`, but the other side is pulling low, we would read back `0x00` instead of what we transmitted.

Though I agree that there should be an option to disable echo if needed (i.e. disable receiver while we are transmitting and enable back once the TX buffer becomes empty). I suggest we do the following:
- add `SERIAL_HALF_DUPLEX` configuration option to enable half-duplex mode in `begin()`
- add `SERIAL_HALF_DUPLEX_NO_ECHO` configuration option to disable echoing
- ensure that when in half-duplex mode, `TX` pins are configured as `alfternate function open drain` instead of `alternate function push-pull` and RX are configured as floating

How does that sound?

ScruffR, 2017-04-14:
Sounds good.
It might be good to check what other patforms have to offer there.

avtolstoy, 2017-04-16:
Half-duplex just means that TX and RX are electrically connected. Hardware-wise you still have 2 separate blocks: receiver and transmitter. It's a perfectly valid configuration when a receiver is enabled while the transmitter is working, which can be useful in some cases (another example I can come up with is collision detection in multi-master configurations). You are not exactly reading what you are transmitting, you are reading back what is seen on the bus, and the data received might differ.

And I'm pretty sure that it's still called 'echo' ;)

ScruffR, 2017-04-17:
If the "echo" works reliably and I do get *all* TX data in the RX buffer, I can work with that :+1:

But the no-echo option is definetly a must have too.

PR #1308 (2017-11-01): USART half-duplex enhancements
### Problem

Original issue: #1276.

### Solution

- Adds `SERIAL_HALF_DUPLEX` configuration option to enable half-duplex mode in `begin()`
- Adds `SERIAL_HALF_DUPLEX_NO_ECHO` configuration option to disable echoing (receiver gets disabled while transmitting)
- default USART half-duplex mode set to push-pull
  - Adds SERIAL_OPEN_DRAIN, SERIAL_HALF_DUPLEX_OPEN_DRAIN and
  - SERIAL_TX_PULL_UP USART config options.

### Steps to Test

`wiring/serial_halfduplex`

### Example App

N/A

### References

---

### Completeness

- [x] User is totes amazing for contributing [... cut]
closed 2018-03-14 by technobly