particle-iot/device-os #1279: Photon stuck on blinking green/cyan - Unfortunate, it's back on 0.6.1

auromabrewingco, 2017-03-22:
I thought this issue was resolved in #804 and #904 already, but I have been getting the same symptoms again many times - using **firmware 0.6.1**

Curious to hear whether others have experienced the same issue after the previous fix @monkbroc @m-mcgowan @technobly 

Please help as our production is blocked due to this! x(

## SETUP

- Photon

- SYSTEM_MODE(AUTOMATIC); [By default]

- SYSTEM_THREAD(ENABLED);

- Firmware 0.6.1

- No Cloud Function called in the loop. Only some variables exposed - which we rarely call.

- No command to connect or disconnect from WiFi or the Particle Cloud.

- The loop is a state machine consisting of 3 simple states. First, pump in water. Second, heat water to 100C. Third, pump out water. Repeat.

## SYMPTOMS

- The issues appeared at random intervals over the past 2 weeks. Sometimes it happened in the middle of the night after running it for a long time. Sometimes it happened 10 minutes after starting the system.

- LED activity was mixed. I've had both blinking green and blinking cyan. [So not the entire system crash]

- When it got stuck, whatever the I/Os were doing at that time continued. For instance, our setup happened to get stuck when the GPIO switching a high-power heater was meant to be ON - so it kept on heating and things got quite nasty. I also lost connection through the Tinker app.

Please let me know if I could provide extra info to help identify and solve the problem.

technobly, 2017-03-22:
@auromabrewingco thanks for the report.  It kind of sounds like it might be stimulated by power or brief internet outages.  I'd be curious to know if you can force the issue to happen with your devices by:

1. Pulling the ethernet connection to a router, if you have it separated like that.  Waiting for the connection to restore.
2. Powering off/on the router. Waiting for the connection to restore.
3. Erasing all Wi-Fi credentials and setting it up to a cellphone hotspot, and cycling that off/on.  Waiting for the connection to restore.

These may take several tries depending on your code interaction.

If you can reproduce this, next steps would be to boil the code down to the minimum that still shows the failure mode.  That would help us confirm the issue.

I'm also curious what other sensors and peripherals you are using, or libraries.  Perhaps SPI or I2C for temperature sensing?  It might be worth me taking a quick peak at your firmware as well.  You can email it to us, but it would probably be best to get a support ticket in place as well so we can properly track it.  https://support.particle.io  Please ask for files to be sent to me, Brett.

avtolstoy, 2020-03-03:
This should be resolved in recent releases.

closed 2020-03-03 by avtolstoy