particle-iot/device-os #1281: Particle.publish( ... , WITH_ACK) does not time out during internet loss

technobly, 2017-03-26:
### Expected behavior

When using Particle.publish( ... , WITH_ACK); if the message cannot be sent due to internet loss, the function call should timeout after 30 seconds.  Docs do not state the timeout expected either.

### Observed behavior

The function call blocks until the internet returns, or the Wi-Fi AP is turned off completely.

### Steps to reproduce

WITH_ACK only available using 0.6.1 firmware. Run the test app below, and either pull the ethernet cable between the modem and router, or switch off the internet connection in your router settings if available.

### Test App

```c
SYSTEM_MODE(SEMI_AUTOMATIC);
SYSTEM_THREAD(ENABLED);

const uint32_t PUBLISH_INVERVAL = 2000;
int count = 1;

void setup() {
    Particle.connect();
    Serial.begin();
}

void loop() {
    static uint32_t lastPublish;

    if (millis() - lastPublish > PUBLISH_INVERVAL)
    {
        lastPublish = millis();
        bool success = false;
        success = Particle.publish("publish-with-ack", String(count), PRIVATE, WITH_ACK);
        if (!success) {
            Serial.printlnf("%7d(s): publish failed: %d", millis()/1000, count);
        }
        else {
            Serial.printlnf("%7d(s): publish passed: %d", millis()/1000, count);
        }
        count++;
    }
}
```

### Observed Output

```
    116(s): publish passed: 58
    118(s): publish passed: 59
    120(s): publish passed: 60
    122(s): publish passed: 61
    124(s): publish passed: 62
    126(s): publish passed: 63
    128(s): publish passed: 64
    130(s): publish passed: 65
    152(s): publish failed: 66 // internet disabled here
    316(s): publish failed: 67 // internet restored here
    317(s): publish passed: 68
    318(s): publish passed: 69
    320(s): publish passed: 70
    322(s): publish passed: 71
    324(s): publish passed: 72
    326(s): publish passed: 73
```

### References

https://community.particle.io/t/particle-publish--with-ack-flag-timeout/30698
https://docs.particle.io/reference/firmware/photon/#particle-publish- (docs do not state a timeout either)

---

Completeness:
- [x] Minimum test case added
- [x] Device, system and user firmware versions stated (WITH_ACK only available using 0.6.1)

sergeuz, 2017-03-27:
The issue seems to be fixed in this PR by @avtolstoy: https://github.com/spark/firmware/pull/1269 – it introduces new protocol command, `ProtocolCommands::DISCONNECT`, which invokes all ACK handlers registered by an application with an error, once connection with the cloud is closed. Here is relevant part of the code: https://github.com/spark/firmware/pull/1269/files#diff-aa3d2967dacfc1cdb97a9266105c531eR79.

avtolstoy, 2017-07-07:
Fixed in #1269 and 394de0fee0125b5f74475027448eefbd1e292655, released in 0.7.0-rc.1

closed 2017-07-07 by avtolstoy