particle-iot/device-os #1284: [Core] Software Timers throw error in /freertos4core/src/concurrent_hal.cpp

ScruffR, 2017-03-30:
As discussed in this thread and tried to discuss on slack
https://community.particle.io/t/connecting-a-keypad-over-i2c/23736/18?u=scruffr

technobly, 2017-03-30:
Thanks @ScruffR, would you please update your bug report to use the same format as this one https://github.com/spark/firmware/issues/1281 ?

technobly, 2017-03-31:
Thanks for the extra details @ScruffR!  I quickly attempted to verify this by using that example, and it compiled with 0.5.3, 0.6.0 and 0.6.1 so I can't confirm it.

Would you try something for me?  On the one you created, can you change one line in your example timers.ino (just add a new blank line) save it, and try compiling again.  Does that fix it?  I think there is still a known bug where trying to compile before all of the library files are copied over will result in an error.  If you encounter that you have to change the app again to have it retry including all of the files again.

technobly, 2017-03-31:
Trying my own advice, I tried to copy and compile quickly and it failed.  Then tried to add one line, save.  It still did not compile.  BUT, when I removed the library, added it back to the app, waited a few... then compiled and it worked fine.  So this is likely a bug to report at https://github.com/spark/cloud but I'm sure I've logged this already internally.  I'll link to this issue as well.  Please close if you can confirm this is not a firmware issue, but rather a Web IDE issue.

ScruffR, 2017-03-31:
@technobly, yup that worked, but this way, we'd have to search for another cause/solution for the member in the linked thread.

closed 2017-03-31 by ScruffR