particle-iot/device-os #1285: System.updatesPending() not working properly

ScruffR, 2017-03-30:
As discussed in this thread
https://community.particle.io/t/how-to-notify-application-incase-of-an-incoming-ota-update/30976
and on slack (2017-03-31T18:36Z) 
`System.updatesPending()` is currently of no use.

This code never indicates a pending update (neither for dev nor for prod devices)
```
STARTUP(System.disableUpdates())

void setup() {
  pinMode(D7, OUTPUT);
}

void loop() {  
  if (System.updatesPending())
    digitalWrite(D7, HIGH);
}
```

Tested on Photon 0.6.1 & Electron 0.6.1

---

Completeness:
- [X] Minimum test case added
- [X] Device, system and user firmware versions stated
- [ ] Particle confirmed

technobly, 2017-03-30:
Currently this API only returns `true` while the user handler for `firmware_update_pending` event is running, right before applying an update.  It's not yet smart about the fact that an update in the Cloud is available when the firmware has disabled updates via `System.disableUpdates()`.  This is something we will build out in the future, and is especially needed for Products.

WellBotLLC, 2017-05-11:
For ultra low powered operations, this function is so very necessary. Trying to keep a device online for longer periods of time consumes a lot of power. I would really like a functional way to check for updates on each loop as such that if returned false, I can put the device back to sleep as fast as possible.

dcliff9, 2017-05-11:
For ultra low powered operations, this function is so very necessary. Trying to keep a device online for longer periods of time consumes a lot of power. I would really like a functional way to check for updates on each loop as such that if returned false, I can put the device back to sleep as fast as possible.

ScruffR, 2017-06-22:
Just to add to the list of victims to this issue
https://community.particle.io/t/photon-ota-vs-system-sleep/33954
And myself for a contract work I recently discussed with Dave

ScruffR, 2018-02-12:
An additional reference
https://community.particle.io/t/force-a-product-update/39429?u=scruffr

ScruffR, 2018-03-27:
@technobly, maybe the this isn't really an issue of `System.updatesPending()` or `firmware_update_pending` itself, but rather one of `Particle.disableUpdates()`. One way to circumvent that might be  *not* to disable the ability to recognise a pending update but rather only the download and application of that update.

Once `System.disableUpdates()` only blockst the latter part of the process but still "listens" for an update, the system event and `System.updatesPending()` could report the fact to re-enable updates.

m-mcgowan, 2019-07-19:
The 1.2.1 release connects the `System.disableUpdates()` API with the cloud, and now operates in a more intuitive manner in conjunction with pending updates.

closed 2019-07-19 by m-mcgowan