particle-iot/device-os #1287: setCredentials() should take care of flash wear

kennethlimcp, 2017-04-03:
## Feature/Enhancement Request

Whenever people ask about `WiFi.setCredentials()` in the community, the advice given would be to make sure that it is not called extensively (every boot up).

The firmware should take care of this so that even in the case that the user is not aware of this edge case, they do not end up damaging/wearing the memory storing the wifi credentials.

### Code from the community

```
  WiFiAccessPoint ap[5];
  int found = WiFi.getCredentials(ap, 5);
  bool gotMyMacCreds = false;
  for (int i = 0; i < found; i++) {
    if(strstr(ap[i].ssid, "SomeSSID"))
    {
      gotMyMacCreds = true;
      Serial.println("had creds");
    }
  }
  if(!gotMyMacCreds)
  {
    WiFi.setCredentials("SomeSSID", "SomePassword");
    Serial.println("set creds");
  }
void loop() {

}
```

### References

- https://community.particle.io/t/setting-additional-wifi-credentials/31314/5

ScruffR, 2017-04-04:
I'm not entirely sure, this isn't already taken care of, but people should still be reminded to think before coding ;-)
The main problem is with people who unconditionally `WiFi.clrearCredentials()` just to follow it up with a `WiFi.setCredentials()` and the firmware won't have any way to prevent that, so we will still have to but-in on these cases.

technobly, 2017-04-07:
In general we try not to erase flash if the resulting write will end up with the same data.  What's not clear here is what happens when the same SSID is used for a connection that previously connected.

The case @ScruffR presents is clearly wearing the flash intentionally and could be added to documentation as a warning (oops, meant to add the other green label backlog->documentation)

closed 2019-12-12 by kennethlimcp