particle-iot/device-os #1290: Allow SPI.begin to not initialize a cs pin

Lindenk, 2017-04-06:
## Feature/Enhancement Request

Hello!

Currently, SPI.begin() will initialize a given chip select pin, or the default if none is given. When using more than one SPI device, initialization of each SPI device (including the device's chip select pin) must be handled by the user, however SPI.begin still needs to be called exactly once. This forces the user to call SPI.begin() with a 'sacrificial pin' that can arbitrarily be set to an output and may cause bugs further down the road if the pin is used for something else.

So my request is to allow the initialization of SPIClass instances without initializing a chip select pin. Either that or better support for dealing with multiple SPI devices on the same bus.

technobly, 2017-04-07:
Hi @Lindenk thanks for the feature request.  This would not be too terribly difficult to implement by using a special case for the pin selection like `SPI.begin(NO_PIN)` where NO_PIN = 65535. Implementation details could be different though but I'll mark this as a confirmed enhancement that we would implement some time in the future.

avtolstoy, 2020-03-03:
`PIN_INVALID` may be passed to `SPI.begin()` in current releases.

closed 2020-03-03 by avtolstoy