particle-iot/device-os #1292: Ability to support more then 5 Sockets on Photon/P0/P1

eely22, 2017-04-08:
## Feature/Enhancement Request

Currently, the Photon firmware can only handle connections on up to 5 sockets, and one of them is used by the Photon for the cloud connection, so really only 4 are available to the user.

If the user code attempts to open a 5th socket, the code appears to work but the firmware simply disconnects the cloud connected socket and the Photon goes into blinking cyan state. 

This is especially important for bluz and the gateway, which requires a socket for every bluz DK that wants to connect through a gateway. As there are only 5 available, bluz can only support 3 bluz DK's connected to the cloud at a time, the sockets end up being used as:
- 1 for Photon
- 1 for nrf51822 on the gateway acting as a BLE central (also a Particle device)
- 3 available for DK's

The bluz gateway can handle up to 8 bluz DK's from a Bluetooth LE standpoint, so extending the number of sockets available on the Photon/P1 would directly allow us to increase the capacity of the gateway. This is a major driver for bluz business as the more DK's that can be supported on a gateway, the more cost effective the solution becomes. 

The WICED firmware seems to support up to 10 sockets, it is restricted to 5 on the Photon specifically, so it would seem to be a configuration change to allow it to work.

The current code highlights the problem, once the 5th socket is opened the Photon will begin to blink cyan:
```
#define MAX_SOCKETS 8

TCPClient clients[MAX_SOCKETS];
uint8_t server[] = { 172, 217, 10, 238 }; // Google
int currentSocket = 0;

void setup() {
    Serial.begin(38400);
    Serial.println("Starting!");
}

void loop() {
    if (currentSocket < MAX_SOCKETS) {
        if (clients[currentSocket].connect(server, 80)) {
            Serial.println("connected");
            clients[currentSocket].println("GET /search?q=unicorn HTTP/1.0");
            clients[currentSocket].println("Host: www.google.com");
            clients[currentSocket].println("Content-Length: 0");
            clients[currentSocket].println();
        }
        else {
            Serial.println("connection failed");
        }
        currentSocket++;
        delay(3000);
    }
}
```

sergeuz, 2017-07-05:
Hey @eely22, I've posted a PR which increases maximum number of open sockets on Photon. Any additional testing and feedback are welcomed!

> If the user code attempts to open a 5th socket, the code appears to work but the firmware simply disconnects the cloud connected socket and the Photon goes into blinking cyan state.

We also fixed this weird behavior in 0.7.0-rc.1, so connections established at the application side won't affect the cloud connection anymore.

eely22, 2018-01-24:
@sergeuz Sorry for the late reply here, but I have tested out this PR and it does appear to work. We have extended our gateway from 3 DK devices to 4 DK (so 6 total open sockets on the Photon) and things appear stable. We will continue to ramp up the number of allowed devices and see if we notice any issues, ultimately we plan to use up all available sockets. Thanks so much!

closed 2018-02-05 by m-mcgowan