particle-iot/device-os #1322: Sting x = NULL produces error on 0.6.2-rc.1 following

ScruffR, 2017-05-04:
## Bug Report

### Expected Behavior

Previously this built without problem

### Observed Behavior

With 0.6.2 (and RCs) formerly valid assignement of `String x = NULL` produces this error output
```
n file included from /usr/local/gcc-arm-embedded/arm-none-eabi/include/stdio.h:35:0,
                 from ./inc/application.h:84,
                 from /workspace/cyanbug.cpp:1:
/workspace/cyanbug.cpp:1:12: error: conversion from 'int' to 'String' is ambiguous
 #include "application.h"
            ^
/workspace/cyanbug.cpp:1:12: note: candidates are:
In file included from ../wiring/inc/spark_wiring_stream.h:30:0,
                 from ../wiring/inc/spark_wiring.h:38,
                 from ./inc/application.h:40,
                 from /workspace/cyanbug.cpp:1:
../wiring/inc/spark_wiring_string.h:68:2: note: String::String(const __FlashStringHelper*)
  String(const __FlashStringHelper *pstr);
  ^
../wiring/inc/spark_wiring_string.h:65:2: note: String::String(const char*)
  String(const char *cstr = "");
  ^
```

This poses a breaking change for some contributed libraries as e.g. MQTT

### Test App

```c
String x = NULL;
void setup() {}
void loop() {}
```

### References

First reported here
https://community.particle.io/t/mqtt-throwing-error/32386

m-mcgowan, 2018-02-05:
This was caused by the addition of the `FastStringHelper` overload. This was primarily done for compatibility, when porting Arduino libraries, but I see that the ambiguous overload gives us anything but that!

@sergeuz any thoughts on how we can fix this specific use case?

ScruffR, 2018-10-18:
In connection with this checks like  `if (someString)` which used to compile fine are now throwing an ambiguity error.
```
conversion from 'String' to 'bool' is ambiguou
```
It worked in the past and still works in Arduino, so it should here too.