particle-iot/device-os #1331: Which one the EEPROM.put() write to ?internal flash of stm32 or eeprom of cc3000?

SharePlainCSource, 2017-05-12:
Which one the EEPROM.put() write to ?internal flash of stm32 or eeprom of cc3000?

I am trying to trace EEPROM.put().

At last, I can not find any stm32 internal flash operation function
EEPROM.put() write eeprom[2047] to eeprom.bin file. Am I right??
fopen("eeprom.bin", "rw");
I think fopen() is in /hal/src/lib/Lib_Wiced_RO_FS.a,But I can not find source cod of  Lib_Wiced_RO_FS.a

Accroding to Photon API:
EEPROM.clear();
Calling this function pauses processor execution (including code running in interrupts) for 800ms since no instructions can be fetched from Flash while the Flash controller is busy erasing both EEPROM pages.

Thanks!

rickkas7, 2017-05-12:
Questions are best handled by the [community forums](https://community.particle.io) or [support](https://particle.io/support) rather than Github issues.

Is this for the Spark Core or the Photon?

The emulated EEPROM (127 bytes) on the Core is in the external SPI flash. It's not in the processor or the CC3000.

The emulated EEPROM (2047 bytes) on the Photon and P1 are in the onboard STM32F205RGY6 flash memory.

The source code to WICED is licensed from Cypress (formerly Broadcom) and is not open-source. Only the built binaries for WICED are checked into the firmware Github repository, which is why you can't find the source.

SharePlainCSource, 2017-05-12:
@rickkas7 
Thanks very much! 
It seems that EEPROM.put() does no work well on my Photon,So I think I should trace the source code
I think EEPROM.put() will call flash operation functions of stm32 in the end.Am I right?

But I find out that:
EEPROM.put() call HAL_EEPROM_Put()
And then in ./hal/src/gcc/eeprom_hal.cpp
void HAL_EEPROM_Put(uint32_t index, const void *data, size_t length)
{
        memcpy(eeprom+index, data, length);
        GCC_EEPROM_Flush();
}

void GCC_EEPROM_Flush()
{
        if (eeprom_file.length()) {
                GCC_EEPROM_Save(eeprom_file.c_str());
        }
}

void GCC_EEPROM_Save(const char* filename)
{
        write_file(filename, eeprom, sizeof(eeprom));
}

At last in ./hal/src/gcc/filesystem.cpp 
void write_file(const char* filename, const void* data, size_t length)
{
    char buf[256];
    buf[0] = 0;
    if (rootDir) {
        strcpy(buf, rootDir);
        strcat(buf, "/");
    }
    strcat(buf, filename);
    FILE *f = fopen(buf, "wb");
    if (f!=NULL) {
        length = fwrite(data, 1, length, f);
        INFO("written file %s length %d", buf, length);
        fclose(f);
    }
    else
    {
        throw invalid_argument(string("unable to write file '") + buf + "'");
    }
}

When and Where the Flash operation functions of stm32 will be called?
It's confusing me. Did I make any mistakes?

rickkas7, 2017-05-12:
All of the stuff in hal/src/gcc is not for the Photon, so that's not the code that's being executed.

The EEPROM function are quite reliable now, but it's easy to use it incorrectly. This post has details on using it to store various types of data: 

https://community.particle.io/t/need-explanation-for-eeprom-usage-generally-and-on-photon/21172/2

And at this point, this is much better handled in the community forums than a Github issue.

SharePlainCSource, 2017-05-12:
@rickkas7 Thanks again！I will read the post

closed 2017-05-12 by SharePlainCSource