particle-iot/device-os #1339: Broken build for PLATFORM=electron with MODULAR=n

screaming-cockatoo, 2017-05-23:
## Bug Report

### Expected Behavior

When compiling with arm-none-eabi-gcc toolchain 4.9.3 20150529 and the following options: PLATFORM=electron MODULAR=n DEBUG_BUILD=y USE_SWD_JTAG=y, a firmware binary should be generated containing all modular parts of the firmware combined in one file.

### Observed Behavior

When passing PLATFORM=electron MODULAR=n DEBUG_BUILD=y USE_SWD_JTAG=y, the build fails with the following errors:

```
../build/target/system/platform-10//libsystem.a(system_network.o): In function HAL_NET_notify_can_shutdown':
firmware/system/src/system_network_internal.h:550: multiple definition ofHAL_NET_notify_can_shutdown'
firmware/hal/src/electron/cellular_hal.cpp:41: first defined here

../build/target/system/platform-10//libsystem.a(system_network.o): In function HAL_NET_notify_connected':
firmware/system/src/system_network.cpp:69: multiple definition ofHAL_NET_notify_connected'
../build/target/hal/platform-10//libhal.a(cellular_hal.o):firmware/hal/src/electron/cellular_hal.cpp:20: first defined here

../build/target/system/platform-10//libsystem.a(system_network.o): In function HAL_NET_notify_dhcp':
firmware/system/src/system_network.cpp:84: multiple definition ofHAL_NET_notify_dhcp'
../build/target/hal/platform-10//libhal.a(cellular_hal.o):firmware/hal/src/electron/cellular_hal.cpp:34: first defined here

../build/target/system/platform-10//libsystem.a(system_network.o): In function HAL_NET_notify_disconnected':
firmware/system/src/system_network.cpp:74: multiple definition ofHAL_NET_notify_disconnected'
../build/target/hal/platform-10//libhal.a(cellular_hal.o):firmware/hal/src/electron/cellular_hal.cpp:27: first defined here
```

I have confirmed that these errors also occur on 0.6.0 and 0.6.1 releases as well, and are only apparent when using PLATFORM=electron with MODULAR=n.

The build does succeed when PLATFORM=photon, or with MODULAR=y.

### Steps to Reproduce

To reproduce this issue, build the project with PLATFORM=electron MODULAR=n DEBUG_BUILD=y USE_SWD_JTAG=y

### References

Issue flagged on Particle Community: https://community.particle.io/t/building-debug-builds-error-doc-topic-debugging-with-eclipse/31697/16

Reproduced by 3 members of the community.

---

avtolstoy, 2017-07-07:
Fixed by #1340, released in 0.7.0-rc.1.

closed 2017-07-07 by avtolstoy

justynb, 2017-11-28:
This seems to be broken again in 0.7.0-rc.5.

Working in 0.7.0-rc.4, same errors as above in 0.7.0-rc.5.

PR #1340 (2017-05-30): Fixes monolithic builds
### Problem

Fixes monolithic builds on:
- Electron (mainly broken by https://github.com/spark/firmware/pull/1184) Fixes #1339
- Photon (broken in `feature/photon/wiced-3.7.0-7` branch due to WICED upgrade and an oversight)

### Solution

- Use tls function callbacks on Photon/P1 only with modular builds
- Use cellular HAL NET callbacks on Electron only in modular builds
- Change preprocessor `#if MODULAR_FIRMWARE` checks to `#if defined(MODULAR_FIRMWARE) && MODULAR_FIRMWARE`, to avoid triggering an error with `-Wundef` which is now always enabled
- Link HAL with `Lib_c [... cut]