particle-iot/device-os #1341: wiced_udp_register_callbacks returns UNSUPPORTED

tlangmo, 2017-06-01:
<details>
  <summary><i>submission notes</i></summary>

```
**Important:** Please sanitize/remove any confidential info like usernames, passwords, org names, product names/ids, access tokens, client ids/secrets, or anything else you don't wish to share.

Please choose one of the following templates **Bug Report** or **Feature/Enhancement Request** and delete the other.

You may also delete this submission notes header if you'd like. Thank you for contributing!
```
</details>

## Bug Report

### Expected Behavior
`wiced_udp_register_callbacks` should return 0 for SUCCESS and callback gets called whenever a new udp packet arrives.

### Observed Behavior

The WICED function `wiced_udp_register_callbacks` compiles without problems, but returns code 7  at runtime, which represents `UNSUPPORTED` according to `wiced_result.h`

### Steps to Reproduce
I tested on develop branch. Quickest way to reproduce is to add the a callback directly in 
`socket_hal.cpp` whenever a UDP socket is created.

### Test App
in `socket_hal.cpp`

```
wiced_result_t  _mycb(wiced_udp_socket_t* socket, void* arg) {
	return WICED_SUCCESS;
}
```

```
... in function socket_create()
...
if (protocol==IPPROTO_TCP) {
  wiced_result = wiced_tcp_create_socket(tcp(socket), wiced_wlan_interface(nif));
}
else {
  wiced_result = wiced_udp_create_socket(udp(socket), port, wiced_wlan_interface(nif));
  // register test callback 
  wiced_result_t  rc = wiced_udp_register_callbacks( udp(socket), _mycb, NULL );
  INFO("wiced_udp_register_callbacks, result=%d",(int)rc);
}
...
```

```
UDP myUdp;
void setup() {
  // A minimal test case to reproduce the issue is super helpful!
  myUdp.begin();
}

void loop() {

}
```
---
## Feature/Enhancement Request
A UDP callback for each packet would be tremendously useful for time synchronization as incoming sync packets can be timestamped without constantly polling and with more accuracy. This in turn allows to synchronize several photons within a single WiFi  network with accuracy of several hundred microseconds, or even better.

This kind of synchronization is widely used and published for example at
[Fine-Grained Network Time Synchronization using Reference
Broadcasts](https://www.usenix.org/legacy/event/osdi02/tech/full_papers/elson/elson.pdf)

I am not sure whether lack of his feature is hardware related or can be enabled in software.

m-mcgowan, 2017-07-07:
Thanks for the clear and well thought out issue report!  We definitely see the value of this feature, but unfortunately this is an omission in the WICED SDK running on LwIP.  UDP callbacks are supported  with the NetX networking stack. 

It's not a limitation of the hardware, so possible to fix in firmware in principle. Since you are calling this function, which is not available from application space, I'm guessing you are building your own system firmware? You may be able to workaround this limitation by building against the NetX stack. 

We'll keep this issue open, in case it is addressed later when the WICED upgrades to a newer version of LwIP, such as the 2.0 line that supports IPv6. #392

tlangmo, 2018-01-17:
@m-mcgowan How can I build locally against the NetX stack? It would love to try that out seeing other problems with WiFi robustness e.g. #1461. But I am puzzled on how to build the wiced libraries for the photon firmware myself. I have downloaded the Wiced 3.7 sources from the Cypress page but what should I do next? Thanks

closed 2020-03-03 by avtolstoy