particle-iot/device-os #1347: Delay between time when Photon actually comes offline and time when Particle cloud publishes "offline" event

SergeyKiselev1995, 2017-06-21:
## Bug Report

### Expected Behavior

When Photon device is disconnected from power supply particle cloud triggers it in short time and publishes "device offline" event so webhooks can handle this event

### Observed Behavior

When Photon is unplugged "device offline" event is published only in 15-20 minutes after actual "disconnect" event

### Steps to Reproduce

This behavior can be reproduced by disconnecting Photon device form power supply

m-mcgowan, 2017-06-21:
Thanks for the issue report. Could you please tell us more about the setup you have:
- what version of system firmware is running on your device.  [how to](https://community.particle.io/t/determining-the-version-of-system-firmware-on-a-photon-p1/13324) 
- how reproducible is this? does it happen every time?
- anything unusual about your system firmware, e.g. are you running UDP rather than TCP?

In principle, the cloud expects to hear from devices at regular intervals and will disconnect the device after 15 seconds.

SergeyKiselev1995, 2017-06-21:
Hi. Thank you for response.

- Firmware version is 0.6.2
- I've tested 10 times and in 9 of them delay was 10 and more minutes. In one case delay was just 1 minute.
- There is nothing unusual in system firmware.

JamesHagerman, 2019-05-10:
The behavior around these events has been updated for Photons. This should no longer be an issue.

closed 2019-05-10 by JamesHagerman