particle-iot/device-os #1349: Support multiple pins to wake from System.sleep(wakeUpPin, edgeTriggerMode)

pkourany, 2017-07-04:
### References

[Docs - System.sleep()](https://docs.particle.io/reference/firmware/photon/#sleep-sleep-)

[waking-up-electron-from-sleep-using-pin-and-sms
](https://community.particle.io/t/waking-up-electron-from-sleep-using-pin-and-sms/32932/4)

---

## Feature/Enhancement Request

Add support for multiple (EXTI) interrupts/pins to wake Photon/Electron from `System.sleep(wakeUpPin, edgeTriggerMode)`.  Configuring an external (EXTI) interrupt via attachInterrupt() as per [note from @m-mcgowan](https://community.particle.io/t/photon-sleep-mode-wakeup-on-multiple-interrupts/15368/7) did not provide the (hoped for) additional pin wake functionality.

teckel12, 2017-07-27:
I can confirm that attachInterrupt() does not wake the Photon from System.sleep().

m-mcgowan, 2017-10-05:
It would also be good if the caller can know what woke up the system, a configured interrupt pin, WKP pin, RTC alarm etc.

ScruffR, 2017-12-27:
This might also be worth taking into account
https://github.com/spark/electron/issues/44

If one wanted to wake an Electron/E from deep sleep via SMS/ring (with 3rd party or future Particle SIM) there is currently no way due to the HW limitation.
What would be a possible SW workaroung (if any)?

zhirschtritt, 2018-01-05:
@avtolstoy - I've tried out the multiple wake up pins on an Electron recently and have not had success. Even without specifying multiple pins (using original working code on v0.6.4), the device will not go into sleep mode. Have you had success when using an Electron with this feature branch?

avtolstoy, 2018-01-08:
@zhirschtritt Just double checked once again using the test app (https://github.com/particle-iot/firmware/blob/c53601e153522c1424429ef227f442297c97e031/user/tests/app/sleep_multiple_pins/application.cpp) from the PR in question (#1405) and it does work for me. Could you provide an example app so that I can try to reproduce the issue with the same starting conditions?

As for your issue with 0.6.4, you might possibly be affected by #1176 (fixed in #1259, and released in 0.7.0-rc.1), if threading is enabled.

PR #1405 (2018-01-22): System.sleep(): support for multiple wake up pins
<details>
  <summary><i>submission notes</i></summary>

```
**Important:** Please sanitize/remove any confidential info like usernames, passwords, org names, product names/ids, access tokens, client ids/secrets, or anything else you don't wish to share.

Please Read and Sign the Contributor License Agreement ([Info here](https://github.com/spark/firmware/blob/develop/CONTRIBUTING.md)).

You may also delete this submission notes header if you'd like. Thank you for contributing!
```
</details>

### Problem

Original issue #1349

### Solution

Imp [... cut]
closed 2018-01-22 by m-mcgowan