particle-iot/device-os #1363: GCC 5.4.1 SRAM overflow on PLATFORM=electron

technobly, 2017-07-26:
## Bug Report

### Expected Behavior

- v0.6.2 firmware
- compiling with GCC v5.4.1
- `cd firmware/modules`
- `make clean all -s PLATFORM=electron`
- should compile Tinker app successfully

### Observed Behavior

`error: region 'SRAM' overflowed by 20 bytes`

sergeuz, 2017-07-26:
Some details: https://github.com/spark/firmware/pull/1159#issuecomment-262969953

technobly, 2017-07-26:
I knew there was some info somewhere, thanks @sergeuz

elcojacobs, 2017-08-20:
This also happens on platform P1, see:
https://community.particle.io/t/linker-error-when-trying-to-build-p1-system-modules-on-0-7-0-solved/35439

Plasmarobo, 2018-11-20:
I've observed this issue attempting to build 0.6.4 with gcc 5.3.1 (5_3-2016q1).
In my case the SRAM is overflowed by 3052 (3060 total bytes of SRAM being consumed).

closed 2020-03-03 by avtolstoy