particle-iot/device-os #1368: Photon Monolithic build fails on 0.7.0-rc.3

adouglass, 2017-08-15:
## Bug Report
When building locally against release/0.7.0-rc.3 a non-modular/monolithic build fails with an error.

`make clean all PLATFORM=photon MODULAR=n`

### Expected Behavior

A single binary containing the full system should be built.

### Observed Behavior

A compiler error is reported.

> ../build/target/hal/platform-6/\libhal.a(heap_4_lock.o):(.data.ucHeapEnd+0x0): undefined reference to `link_heap_location_end'
> ../build/target/hal/platform-6/\libhal.a(heap_4_lock.o):(.data.ucHeap+0x0): undefined reference to `link_heap_location'
> collect2.exe: error: ld returned 1 exit status
> make: *** [../build/target/main/platform-6/main.elf] Error 1

### Steps to Reproduce

* arm-none-eabi-gcc (GNU Tools for ARM Embedded Processors) 5.3.1 20160307 (release) [ARM/embedded-5-branch revision 234589]
* Windows 10

* `git checkout release/v0.7.0-rc.3`
* `make clean all PLATFORM=photon MODULAR=n`

technobly, 2017-08-15:
Introduced in 0.7.0-rc.2

ynezz, 2017-08-29:
Hi!

I'm just trying out the latest RC with this fix:

```shell
git log -10 --pretty=oneline spark/release/v0.7.0-rc.3..
c9b671701a1a36c47ed026f1ff92d5b239141277 link_heap_location and link_heap_location_end definitions missing
```
then trying out following command `make clean all PLATFORM=photon MODULAR=n` which ends with this error:

```
Building c file: src/photon/core_hal.c
Invoking: ARM GCC C Compiler
mkdir -p ../build/target/hal/platform-6/./src/photon/
arm-none-eabi-gcc -DSTM32_DEVICE -DSTM32F2XX -DPLATFORM_THREADING=1 -DPLATFORM_ID=6 -DPLATFORM_NAME=photon -DUSBD_VID_SPARK=0x2B04 -DUSBD_PID_DFU=0xD006 -DUSBD_PID_CDC=0xC006 -g3 -gdwarf-2 -Os -mcpu=cortex-m3 -mthumb -DUSE_STDPERIPH_DRIVER -DDFU_BUILD_ENABLE -fno-builtin-memcpy -fno-builtin-memcmp -fno-builtin-memset -DMBEDTLS_CONFIG_FILE="<mbedtls_config.h>" -DPRODUCT_ID=6 -DPRODUCT_FIRMWARE_VERSION=65535 -DSYSTEM_VERSION_STRING=0.7.0-rc.3 -DRELEASE_BUILD -Werror -I./inc -I./shared -I./src/photon -I./src/stm32f2xx -I./src/stm32 -I./src/photon/api -I./src/photon/include -I./src/photon/wiced/security/BESL/host/WICED/ -I./src/photon/wiced/security/BESL/include -I./src/photon/wiced/security/BESL -I./src/photon/wiced/security/BESL/crypto -I./src/photon/wiced/WWD/include/ -I./src/photon/wiced/platform/include/ -I./src/photon/wiced/platform/GCC/ -I./src/photon/wiced/security/BESL/supplicant/ -I./src/photon/libraries/crypto -I../platform/shared/inc -I../platform/MCU/STM32F2xx/CMSIS/Include -I../platform/MCU/STM32F2xx/CMSIS/Device/ST/Include -I../platform/MCU/STM32F2xx/SPARK_Firmware_Driver/inc -I../platform/MCU/shared/STM32/inc -I../platform/MCU/STM32F2xx/STM32_StdPeriph_Driver/inc -I../platform/MCU/STM32F2xx/STM32_USB_Device_Driver/inc -I../platform/MCU/STM32F2xx/STM32_USB_Host_Driver/inc -I../platform/MCU/STM32F2xx/STM32_USB_OTG_Driver/inc -I../services/inc -I../dynalib/inc -I../bootloader/src/photon -I../bootloader/../hal/src/photon/wiced/platform/MCU -I../bootloader/../hal/src/photon/wiced/platform/MCU/STM32F2xx/WAF -I../bootloader/src/stm32f2xx -I../wiring/inc -I../crypto/inc -I../crypto/mbedtls/include -I../communication/src -I./src/photon/include -I./src/photon/libraries/daemons/DNS_redirect -I./src/photon/libraries/daemons/HTTP_server -I./src/photon/libraries/protocols/DNS -I./src/photon/libraries/utilities/linked_list -I./src/photon/libraries/utilities/ring_buffer -I./src/photon/platforms/BCM9WCDUSI09 -I./src/photon/wiced -I./src/photon/wiced/RTOS/FreeRTOS -I./src/photon/wiced/RTOS/FreeRTOS/WICED -I./src/photon/wiced/RTOS/FreeRTOS/WWD -I./src/photon/wiced/RTOS/FreeRTOS/WWD/ARM_CM3 -I./src/photon/wiced/RTOS/FreeRTOS/ver8.2.1/Source/include -I./src/photon/wiced/RTOS/FreeRTOS/ver8.2.1/Source/portable/GCC/ARM_CM3 -I./src/photon/wiced/WWD -I./src/photon/wiced/internal -I./src/photon/wiced/network/LwIP -I./src/photon/wiced/network/LwIP/WICED -I./src/photon/wiced/network/LwIP/WWD -I./src/photon/wiced/network/LwIP/WWD/FreeRTOS -I./src/photon/wiced/network/LwIP/ver1.4.1/src/include -I./src/photon/wiced/network/LwIP/ver1.4.1/src/include/ipv4 -I./src/photon/wiced/platform/ARM_CM3 -I./src/photon/wiced/platform/ARM_CM3/CMSIS -I./src/photon/wiced/platform/GCC -I./src/photon/wiced/platform/MCU -I./src/photon/wiced/platform/MCU/STM32F2xx -I./src/photon/wiced/platform/MCU/STM32F2xx/WAF -I./src/photon/wiced/platform/MCU/STM32F2xx/peripherals -I./src/photon/wiced/platform/include -I./src/photon/wiced/security/BESL/crypto/ -I./src/photon/wiced/security/BESL/crypto/ -I./src/photon/wiced/security/BESL/crypto_open/ -I./src/photon/wiced/security/BESL/crypto_open/ -I./src/photon/wiced/security/BESL/crypto_open/ -I./src/photon/wiced/security/BESL/crypto_open/ -I./src/photon/wiced/security/BESL/crypto_open/ -I./src/photon/wiced/security/BESL/crypto_open/ -I./src/photon/wiced/security/BESL/crypto_open/ -I./src/photon/wiced/security/BESL/crypto_open/ -I./src/photon/wiced/security/BESL/crypto_open/ -I./src/photon/wiced/security/BESL/crypto_open/ -I./src/photon/wiced/security/BESL/crypto_open/ -I./src/photon/wiced/security/BESL/crypto_open/ -I./src/photon/wiced/security/BESL/crypto_open/ -I./src/photon/wiced/security/BESL/crypto_open/ -I./src/photon/wiced/security/BESL/crypto_open/ -I./src/photon/wiced/security/BESL/crypto_open/ -I./src/photon/wiced/security/BESL/crypto_open/ -I./src/photon/wiced/security/BESL/crypto_open/ -I./src/photon/wiced/security/BESL/crypto_open/ -I./src/photon/wiced/security/BESL/crypto_open/ -I./src/photon/wiced/security/BESL/crypto_open/ -I./src/photon/wiced/security/BESL/crypto_open/ -I./src/photon/wiced/security/BESL/crypto_open/ed25519/ -I./src/photon/wiced/security/BESL/crypto_open/ed25519/ -I./src/photon/wiced/security/BESL/crypto_open/ed25519/ -I./src/photon/wiced/security/BESL/crypto_open/ed25519/ -I./src/photon/wiced/security/BESL/crypto_open/ed25519/ -I./src/photon/wiced/security/BESL/crypto_open/ed25519/ -I./src/photon/wiced/security/BESL/crypto_open/ed25519/ -I./src/photon/wiced/security/BESL/crypto_open/ed25519/ -I./src/photon/wiced/security/BESL/crypto_open/ed25519/ -I./src/photon/wiced/security/BESL/crypto_open/ed25519/ -I./src/photon/wiced/security/BESL/crypto_open/ed25519/ -I./src/photon/wiced/security/BESL/crypto_open/ed25519/ -I./src/photon/wiced/security/BESL/crypto_open/ed25519/ -I./src/photon/wiced/security/BESL/crypto_open/ed25519/ -I./src/photon/wiced/security/BESL/crypto_open/ed25519/ -I./src/photon/wiced/security/BESL/crypto_open/unit/ -I./src/photon/wiced/security/BESL/DTLS/ -I./src/photon/wiced/security/BESL/DTLS/ -I. [... cut ...]
../platform/MCU/STM32F2xx/SPARK_Firmware_Driver/inc/hw_config.h:87:5: error: "MODULE_FUNCTION" is not defined [-Werror=undef]
 #if MODULE_FUNCTION != MOD_FUNC_BOOTLOADER
     ^~~~~~~~~~~~~~~
cc1: all warnings being treated as errors
```

```
arm-none-eabi-gcc (GNU Tools for ARM Embedded Processors 6-2017-q2-update) 6.3.1 20170620 (release) [ARM/embedded-6-branch revision 249437]
Copyright (C) 2016 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

Compiler version (un)related issue?

ynezz, 2017-08-29:
I get exactly the same error even with older GCC:

```
arm-none-eabi-gcc (GNU Tools for ARM Embedded Processors) 5.4.1 20160919 (release) [ARM/embedded-5-branch revision 240496]
Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

```
In file included from ./src/stm32f2xx/core_hal_stm32f2xx.h:4:0,
                 from src/photon/core_hal.c:27:
../platform/MCU/STM32F2xx/SPARK_Firmware_Driver/inc/hw_config.h:87:5: error: "MODULE_FUNCTION" is not defined [-Werror=undef]
 #if MODULE_FUNCTION != MOD_FUNC_BOOTLOADER
     ^
cc1: all warnings being treated as errors
make[1]: *** [../build/target/hal/platform-6/./src/photon/core_hal.o] Error 1
```

technobly, 2017-08-29:
Hi @ynezz!  Please use GCC version 5.3.1 with 0.7.0 for now.  Version 5.4.1 and 6.x.x are currently not supported.  Also please make sure you have the latest version of the branch:
`git checkout release/v0.7.0-rc.3 && git fetch origin && git reset --hard origin/release/v0.7.0-rc.3`

technobly, 2017-08-29:
@ynezz @avtolstoy spotted that you might be building in the `firmware $` directory, and should be building in the `firmware/main $` directory ;-)

technobly, 2017-08-29:
Gah, also this is still currently broken in `release/v0.7.0-rc.3`.  You need to checkout this branch for the fix `fix/0.7.0-rc.3-monolithic-build`

ynezz, 2017-08-29:
Thanks! I'm able to build `release/v0.7.0-rc.3` with cherry-picked `92095f75ebdbbe076932a83b48f92f1084107463` just fine with GCC version 5.3.1.

PR #1370 (2017-11-06): Fixes Photon monolithic builds on 0.7.0-rc.3
### Problem

See #1368 

`link_heap_location` and `link_heap_location_end` definitions missing in linker file.

### Solution

Cherry-pick a71eba6aed768356a2d696e73fa28aa8ef158eea from #1369

### Steps to Test

Monolithic build should now succeed.

### Example App

N/A

### References

Closes #1368

---

### Completeness

- [x] User is totes amazing for contributing!
- [x] Contributor has signed CLA ([Info here](https://github.com/spark/firmware/blob/develop/CONTRIBUTING.md))
- [x] Problem and Solution clearly stated
- [x] Run unit/integration/applic [... cut]
closed 2018-03-14 by technobly