particle-iot/device-os #1379: Add pulseIn variant with timeout

monkbroc, 2017-08-30:
Arduino has a variant of `pulseIn` with a 3rd argument `timeout`. Particle only has the 2 argument version. This is breaking Arduino libraries and examples that use that 3 argument variant.

`pulseIn(pin, value, timeout)`

https://www.arduino.cc/en/Reference/PulseIn

Current implementation:

https://github.com/spark/firmware/blob/fc368898d796921994cffcc9d7cc5320e41281a8/wiring/inc/spark_wiring.h#L79

rkast, 2020-07-20:
I have an application that would require this enhancement to be implemented. Any chance there's been progress?

rkast, 2020-07-20:
I've created a pull request with an update to implement this enhancement:

https://github.com/particle-iot/device-os/pull/2158