particle-iot/device-os #1381: SPISettings gone missing with 0.6.2-rc.1 (and all following)

ScruffR, 2017-09-06:
I just tried to recompile an old sketch which was using `SPISettings()` and worked woth 0.6.1, but with 0.7.0-rc.3 I got 
```
'SPISettings' was not declared in this scope
```
And after backtracking I found that 0.6.1 (including RCs) was the first ***and*** last version that supported that.

m-mcgowan, 2017-09-06:
This is by design. SPISettings were added late in the game, which broke some libraries and applications that were already defining them. 0.6.2-rc.1 introduced Arduino compatibility via the Arduino header, along with the `ARDUINO` version define that library authors use to determine which arduino features are available.

ScruffR, 2017-09-06:
In that case the docs are omitting this crucial information tho'

technobly, 2017-09-06:
@ScruffR how does this sound? https://github.com/spark/docs/issues/681

ScruffR, 2017-09-07:
Yup, that may do (more see in spark/docs#681)

m-mcgowan, 2017-11-24:
Can we close this one now?

ScruffR, 2017-11-24:
I'm still not completely convinced, but will have to put up with it.

closed 2017-11-24 by ScruffR