particle-iot/device-os #1392: Photon stuck while blinking rapidly green

JFdeB, 2017-09-20:
<details>
  <summary><i>Photon rapidly blinking green.
the photon is blinking rapidly green and unresponsive, program is stuck.
The photon is supposed to do a small task and go to sleep for 10 minutes. Battery will last for months, but when the photon gets stuck the battery drains in a day. This happens after the program has run for weeks.</i></summary>

Photon rapidly blinking green.
the photon is blinking rapidly green and unresponsive, program is stuck.
The photon is supposed to do a small task and go to sleep for 10 minutes. Battery will last for months, but when the photon gets stuck the battery drains in a day. This happens after the program has run for weeks.
The problem started when i flashed the 0.6.1 firmware.
My code is below, the main problem is that there is no time out on connecting the WIFI. If that were the case, the execution of the loop would fail, photon would go to sleep and try again the next time.
CAN you check the WIFI connect code and make sure the timeout is always working?

SYSTEM_THREAD(ENABLED);
STARTUP(WiFi.selectAntenna(ANT_INTERNAL));

#include "Adafruit_BME280/Adafruit_BME280.h"
#define SEALEVELPRESSURE_HPA (1013.25)
#include "PowerShield/PowerShield.h"

PowerShield batteryMonitor;

void setup() {
    
}

void loop() {
    //    float temperature;
    String messageString;
    Adafruit_BME280 bme;
    if ( !Wire.isEnabled() ) {
        Wire.begin();
    }
    //    Wire.begin();
    delay(1000);
    
    batteryMonitor.reset();
    batteryMonitor.quickStart();
    
    bme.begin();
    
    delay(1000);

    float cellVoltage = batteryMonitor.getVCell();
    float stateOfCharge = batteryMonitor.getSoC();
    float bme_t = bme.readTemperature();
    float bme_h = bme.readHumidity();
    float bme_p = bme.readPressure() / 100.0F;
    
//    WiFi.on();
    waitFor(Particle.connected, 30000);
    if (Particle.connected()) {
        //    messageString = String(cellVoltage) +',' + ' ' + String(stateOfCharge);
        //    Particle.publish("ps-soc", messageString, 60, PRIVATE);
        bme_t = bme.readTemperature();
        bme_h = bme.readHumidity();
        bme_p = bme.readPressure() / 100.0F;
        messageString = String(cellVoltage) +',' + ' ' + String(stateOfCharge) +',' + ' ' + String(bme_t) + ',' + ' ' + String(bme_h) + ',' + ' ' + String(bme_p) + ',' + ' ' + String(WiFi.RSSI()) + ',' + ' ' + String(WiFi.SSID());
        // messagestr = cellVoltage, stateOfCharge, mcp_t, mcp2_t, bme_t, bme_h, bme_p, dht_t, dht_h, WiFi_RSSI, WiFi_SSID
        Particle.publish("temp-hum", messageString, 60, PRIVATE);
        //Particle.publish("acknowledge", messageString, PRIVATE, WITH_ACK);
    }
    
    delay(5000); //extra delay added after messages were not revceived March 23, 2017. Problem of v0.6.1
    
    //  loop takes 7 seconds
    //    System.sleep(SLEEP_MODE_DEEP, 53);
    //    delay(53000);
    System.sleep(SLEEP_MODE_DEEP, 593);
    
}```
**Important:** Please sanitize/remove any confidential info like usernames, passwords, org names, product names/ids, access tokens, client ids/secrets, or anything else you don't wish to share.

Please choose one of the following templates **Bug Report** or **Feature/Enhancement Request** and delete the other.

You may also delete this submission notes header if you'd like. Thank you for contributing!
```
</details>

## Bug Report

### Expected Behavior

What should it do?
When the WIFI connection fails, the code should come out with a timeout
Timeout does not seem to work

### Observed Behavior

What did it actually do?
The program gets stuck on connecting the wifi

### Steps to Reproduce

How did you make it do that, and which system/user firmware versions were tested?
Firmware 0.6.1
Problem reproduces every few weeks at random

### Test App
See above
```c
void setup() {
  // A minimal test case to reproduce the issue is super helpful!
}

void loop() {

}
```

### References

Links to the Community, Docs, Other Issues, etc..

---

## Feature/Enhancement Request

Please describe your use case in as much detail as possible. Background information and detailing what the feature will enable you to do, is especially helpful.

gitjoost, 2017-12-07:
I am not familiar with the battery monitor Powershield nor Adafruit_BME280, however I couldn't help noticing that the first part of your loop() contains initialization statements that should be in setup(). I suggest you put from "String messageString" to "bme.begin()" in setup().    I cannot tell what your problem is specifically, random fails are notoriously difficult to troubleshoot. But you are constantly redeclaring and initializing thereby doing all sorts of things on the stack. That would be my first great suspicion for a random crash. (Particularly the String class, I avoid it like the plague).

ScruffR, 2018-01-31:
@JFdeB, I'm not sure this qualifies as a real firmware issue but mainly rooted in the way how the code is written. Requests like this usually are better handled in the Particle forum.

I'd propose to close this issue.

closed 2020-03-03 by avtolstoy