particle-iot/device-os #1399: Particle.connect() hard blocking since 0.6.1-rc.1

ScruffR, 2017-09-27:
As discussed in [this post](https://community.particle.io/t/electron-firmware-alerting-depending-on-a-variable/36206/9?u=scruffr) the behaviour of `Particle.connect()` has changed with 0.6.1-rc.1 to completely block in non-AUTOMATIC modes without `SYSTEM_THREAD(ENABLED)`.
Before `Particle.connect()` just set a flag and then left to continue running user code (considerably blocked by "background" tasks, but running).
Now the code following `Particle.connect()` does not get executed ***at all*** till the connection is established.

I'd consider this a silently breaking change, since the new "misbehaviour" only hits under certain conditions.

```cpp
SYSTEM_MODE(SEMI_AUTOMATIC);

void setup() 
{
    pinMode(D7, OUTPUT);
    Particle.connect();

    while(1)
    {
      digitalWrite(D7, !digitalRead(D7));
      delay(100);
    }
}
```
Above code does immediately flash the LED on bootup and occasionally during connectino attempts pre 0.6.1-rc.1 but with 0.6.1-rc.1 starting it never blinks till the connection has been established.
See desired behaviour with 0.6.0 https://youtu.be/YWVmkq3Z-mI

technobly, 2017-09-27:
I believe this was a bug that was fixed.

See docs for expected behavior: https://docs.particle.io/reference/firmware/electron/#semi-automatic-mode

https://github.com/spark/firmware/issues/973

ScruffR, 2017-09-27:
Hmm, I'd rather go with Mat's first inclination that the original behaviour is not a bug and the use of `waitUntil()`/`waitFor()` would help the issue. 

But I also did some - not as extensive - tests with MANUAL mode and for that I think the behaviour is wrong. It's hard blocking there too despite Mat's intention to have it non-blocking for MANUAL mode.

I'll do some more testing tho'

technobly, 2017-09-27:
Is there some reason you cannot use `SYSTEM_THREAD(ENABLED);` along with or without `waitUntil() / waitFor()` to get the use case you desire?  Please close the issue if you feel it is sufficiently resolved, thanks!

ScruffR, 2017-09-28:
No real reason, but a sight feeling that there is no perfect solution either way.

m-mcgowan, 2017-09-28:
I am re-opening this since the PR that was referenced above needs reworking slightly to align with the expected behaviour as described in the docs, namely that `Particle.connect()` does not block and that the blocking happens next time `setup()`, `loop()` are called.  Open question is if this should be extended to `delay()` also.

elcojacobs, 2017-12-28:
I'd like to add a note to this issue that in my tests with RC-6 and SYSTEM_THREAD(ENABLED) WiFi.connect() is BLOCKING!

This means that if WiFi goes down, it completely stops my application and the photon spends all of its time trying to reconnect to WiFi. All temperature control stops and my customer's beer is in danger of being ruined!

How do I get the old behavior back of actually handling this in the system thread?

elcojacobs, 2018-01-04:
Upon further investigation, it seems that WiFi.connect() is not blocking directly, but it blocks execution of loop() while WiFi is connecting (with system thread enabled).

Could that be explained by how it is implemented currently and is there a workaround?

aeris-ming, 2018-01-30:
@elcojacobs  I have the same issue.When the internet connect is not stable or wifi signal is weak,the user thread will block forever.Have you find any way to solve this problem?

aeris-ming, 2018-01-30:
@ScruffR @m-mcgowan @technobly If any of your guys find a solution of this issue.Please Please tell me,Thanks very much! WIFI connection is not so important for me compare to keep device run stable,Please!

technobly, 2018-01-30:
@elcojacobs @aeris-ming This thread is for Particle.connect(), not WiFi.connect(), so I wonder if you have identified something different.  If you have the ability to compile locally, would you see if this PR (slated for 0.7.0-rc.7) fixes your issue? https://github.com/particle-iot/firmware/pull/1403

aeris-ming, 2018-03-07:
@elcojacobs https://github.com/particle-iot/firmware/issues/1486
https://github.com/particle-iot/firmware/issues/1449 Do you have any idea why user loop blocked when wifi signal is weak?I have report these issue months ago.

aeris-ming, 2018-03-07:
@elcojacobs https://github.com/particle-iot/firmware/issues/1407 If you have any idea to fix this issue,Please tell me,Thanks!

aeris-ming, 2018-03-19:
@elcojacobs 
Can you search your project user part to see if your have call any sync-system-function?
They may block forever.
@m-mcgowan Help me find out my mistake, I called the sync-system-function hasCredentials() which may block forever.
https://docs.particle.io/reference/firmware/photon/#system-functions

jasongarner, 2018-06-11:
https://community.particle.io/t/particle-connect-timeout-retrydelay/41506/6

> I like the idea of the user firmware having the ability to know when connections can’t be made. I think it would help devices with connection issues function, like how desktop apps have something like an offline mode.

jasongarner, 2018-07-25:
https://community.particle.io/t/particle-connect-timeout-retrydelay/41506

technobly, 2018-07-26:
>Another (potential) instance where users might not anticipate the side-effect of blocking Particle.connect() in SYSTEM_THREAD(ENABLED) mode.

I ran into this myself yesterday :(  

We need to spend some time creating an easy to digest chart of system modes, threaded, non-threaded, with what blocks and what doesn't.

In my opinion `loop()` should always run in a threaded app.

closed 2020-03-03 by avtolstoy