particle-iot/device-os #1404: Switching between multiple known WiFi networks via the code

timpron, 2017-10-11:
<details>
  <summary><i>submission notes</i></summary>

```
**Important:** Please sanitize/remove any confidential info like usernames, passwords, org names, product names/ids, access tokens, client ids/secrets, or anything else you don't wish to share.

Please choose one of the following templates **Bug Report** or **Feature/Enhancement Request** and delete the other.

You may also delete this submission notes header if you'd like. Thank you for contributing!
```
</details>

## Feature/Enhancement Request

I am trying to make photon work with multiple networks, each for a different purpose, for my project. Say I could use WiFi A as my main network and use WiFi B as a backup network or use this network to restart the power supply of WiFi A, essentially restarting WiFi A, whenever WiFi A loses internet connection or WiFi A is down for some reason. As a backup network, WiFi B is always powered, meaning photon is able to detect both networks.

I imagine the pseudo-code can be something like:

if (WiFi A loses connection){
  WiFi.disconnect();
  WiFi.connect(WiFiB, passwordB);
  //then with a peripheral connected to WiFi B, restart the power for WiFi A
  WiFi.disconnect();
  WiFi.connect(WiFiA, passwordA);
}
However, WiFi.connect() is not made to be used that way. Instead, we have WiFi.setCredentials(SSID, PASSWORD); and a separate WiFi.connect(). And, also, WiFi.connect(), in the presence of multiple available networks chooses the connection with the strongest RSSI, and not necessarily as preferred by the user. (https://community.particle.io/t/multiple-wvlans/10720/17?u=engr.pron)

One approach I've seen here in https://community.particle.io/t/switching-between-wifi-devices-solved/17388 makes use of clearing and setting the credentials of the WiFi Networks. In the same thread, however, ...

According to ScruffR:
Switching between Wifi devices [Solved]
"You need to be aware that the WiFi credentials are stored in flash without good wear leveling, so if you keep clearing and setting credentials e.g. in loop() you might wear out your flash, rendering your device unusable."

I would like to suggest then a feature that would best allow switching of connections between multiple known networks via the code. (I see that this is possible with listening mode in tandem with the mobile app, but as it is for the purpose of a backup network, I believe it is best accomplished as an automatic feature - written in the firmware and no user interaction required)

The pseudo-code above with a different functionality for WiFi.connect() might be a good suggestion.

Thanks in advance!

m-mcgowan, 2017-10-19:
Thanks for the clear write up and use case.  I can definitely see why this might be needed.

How about this for a proposal:

- Rather than changing the WiFi.connect() call, since this is often called by the system in AUTOMATIC mode, we instead introduce a filter for the SSIDs that the device will connect to. 

```
WiFi.select("WiFi-A");
```

On the next connection, this will restrict connecting to SSIDs that do not match the filter, i.e. will only connect to SSIDs starting with `WiFi-A`.

The `WiFi.select()` call will not be stored in flash (so no flash wear concerns) and will function until the next device reset. 

How does that sound?

timpron, 2017-10-23:
Okay. From what I can understand, WiFi.connect() is often called by the system in automatic mode. So the idea is to filter out to which network does WiFi.connect() connect to whenever called. Sounds good to me.

So the code would instead be:

if (WiFi A loses connection){
WiFi.disconnect();
WiFi.select (WiFiB);
WiFi.connect();
//then with a peripheral connected to WiFi B, restart the power for WiFi A
WiFi.disconnect();
WiFi.select (WiFiA);
WiFi.connect();
}

But wait. Can we not "restrict" connecting to SSIDs that do not match the filter? Instead, prioritize connecting to SSIDs that match the filter. I'm thinking of using the WiFi.select("preferred network") on setup(). Like call it before WiFi.connect() so that it connects to the preferred network. However, if in case, I entered the wrong SSID spelling or the said network is actually unavailable upon my call, it would be good if WiFi.connect() will fallback to my other registered SSIDs.

I do not really know the intricacies of your code. So I'm kind of insensitive as to making code revision requests. Haha. Anyways, I hope my suggestion, at least, makes sense.

ScruffR, 2017-12-04:
Additionally a way to just connect to the _stronges_ known WiFi nearby would be a nice feature.

jenschr, 2017-12-04:
I second @ScruffR's suggestion of connecting to the strongest SSID. Alternatively one could also have another overload of the select.method:

```
> byte myBSSID[6];
> WiFi.select(myBSSID);
```

That would yield an even more more powerful solution that will make it possible to create a good solution for those with more than a single router on the same SSID name (for extending range). Right now, the WICED firmware seems to connect to the last connected router, not the strongest. In workplaces it's not very popular to run around and restart the routers and this would enable us to make very smooth solutions for the end user.

badslug, 2018-04-12:
We really need the ability to set the BSSID. As @jenschr suggests, allowing a selection of BSSID would be ideal. Our deployments frequently have 3+ APs visible with the same SSID at any time and it's critical to reliable connectivity that we can control the specific base station connections are made to.

In the forums, mdma mentions that the WICED firmware settings has been moved out into Particle hal code: https://community.particle.io/t/connect-to-specific-bssid/37845/8

It seems that the join call passes in `ap.details` which contains a BSSID field. Does anyone know if the WICED firmware looks at the BSSID when it joins?

https://github.com/particle-iot/firmware/blob/e1e856465217923b734e8c40a6f8fde55fd60e6b/hal/src/photon/wlan_hal.cpp#L621

gmantx, 2018-04-13:
We deploy the networking infrastructure for our customers but use a generic image/config so that we can repeat and easily deploy and support without having to customize the basic SSID and associated credentials.  The deployment then ends up having multiple APs with the same configuration that provides a nice seamless architecture.  

In our current embedded environment (deprecated due to module EOL and thus our shift to Particle) we wrote firmware routines to evaluate and then select the best AP based on current status.  It works great and we have seen a dramatic increase in the stability and reliability of our field deployments.  The ability to evaluate and select (using BSSID MAC) via Particle APIs would be the perfect solution.

hramadan, 2019-08-04:
@jenschr's comment:

> the WICED firmware seems to connect to the **last connected** router, **not the strongest**.

is a definite worry in WLAN environments with multi APs using the same SSIDs, which is common in corporate settings. 

I think the solution is to select the AP with the **strongest SSID** as suggested by @ScruffR as it makes logical sense and would be an automatic operation. 

IMHO having an API to select the BSID is a bit kludgy. I would rate it as a work around rather than a robust solution. 

In relation to @timpron's originating post, am thinking that a different approach to the multiple available networks (ie different SSIDs) issue could be to set a **connection priority** against the stored WiFi credentials table. 

Currently the list is traversed in LIFO order that credentials where added (I think). 

In @timpron's example, the solution could reduce to 
` 
  WiFi A set to priority 0 /* highest */
  WiFi B set to priority 4 /* lowest */`

So, if WiFi A is not available, it immediately checks WiFi B (if there are no other WiFi credentials stored with higher priority). This allows the user to add other back up on the spot without code change - say add WiFi C with priority 1. 

This approach gives one flexibility at *configuration time* (which makes sense) rather than via application manipulation at some future date.

causalloop, 2019-08-31:
Just to add another use case - lots of similar things, but with some minor editing from the [forum post ](https://community.particle.io/t/changing-wifi-network-to-another-stored-network-based-on-x-thing/51954)I made:

> For reasons I’ll go into later if anyone is curious, I have an Argon connecting to a little mifi hotspot in a vehicle. When I get home and park, in certain cases I’ll want to switch networks from the mifi to my home wifi. For the most part I don’t really care, but in this particular case, I wanted to have the Argon talk to something using a specific IP address (I don’t want to port forward as the task is intended to be very simple and will not have much in the way of security).
> 
> I have both networks’ credentials stored and I believe the home network is first in the order, so if I reboot it, sometimes it’ll connect to the home wifi - but not always. I can speculate as to why, but rather than go into that part, I want to start with what should be an easy solution: have a cloud function to toggle between the two. I looked through the docs and it looks like WiFi.connect() only has one optional parameter. I’d like to tell it to (effectively) do: WiFi.connect(“homewifi”) etc. I saw someone mentioning clearing the credentials, but a follow up mentioned flash wear and I really have no idea how often I’ll be doing this - might be a couple times, might be a zillion if I come up with some other use case.