particle-iot/device-os #1414: How to generate DCT(16k)  for Photon?

aeris-ming, 2017-10-26:
I want to generate DCT for Photon so that I can flash whole firmware (bootloader dct system-firmware...) by  st-link or j-link.
1.I have read DCT layout part in datasheet.
https://github.com/spark/firmware/blob/release/stable/platform/MCU/STM32F2xx/SPARK_Firmware_Driver/inc/dct.h
2.Read whole stm32 flash data out by st-link.
3.But It seems that It is not simple write application_dct_t struct to flash 0x8004000 or 0x800800.
4.I can not find src of function "dct_write_app_data" from system-firmware.

particle and dfu-util always fail and we pay works in minutes.

Please Please help!Thanks!

m-mcgowan, 2017-10-26:
- to build the DCT, flash the bootloader and system firmware to an erased device
- erase sectors 0x8004000 and 0x8008000 to be certain they are clear
- enter DFU mode
- flash the server public key via DFU (`dfu -d 2b04:d006 -a 1 -s 2082 -D cloud_public_key.der`)
- make any other DCT changes via DFU
- use jtag to save the region 0x8004000-0x8020000 - this is the DCT
- alternatively, you could save the entire region of 0x8000000-0x81000000 which contains everything

During this process it is important that the device remains in DFU mode or is stopped (via jtag). As soon as the device boots, it will attempt to generate a private key and you don't want this to be duplicated across all devices.

aeris-ming, 2017-10-27:
@m-mcgowan Thanks! 
I want to generate DCT with Photon Private Key and my own SSID prefix and suffix.
1.After read the bin file read out by st-link,I find that SSID prefix of  "application_dct_t" is at 0x400a490,"application_dct_t" struct is not start from 0x8004000 or 0x8008000
2.Only application_dct_t struct in DCT part?How to save a application_dct_t as DCT format?

aeris-ming, 2017-10-27:
@m-mcgowan 
It seems "application_dct_t" write to flash (0x8004000 + 0x1D7C) or (0x8008000 + 0x1D7C).
What is the other part of DCT except "application_dct_t" struct?

hal/src/photon/lib/ThreadX/STM32F2XX.s

Disassembly of section .text.dct_write_app_data:
00000000 <dct_write_app_data>:
   0:   4603            mov     r3, r0
   2:   f501 50eb       add.w   r0, r1, #7520   ; 0x1d60
   6:   301c            adds    r0, #28
   8:   4619            mov     r1, r3
   a:   f7ff bffe       b.w     0 <dct_write_app_data>

aeris-ming, 2017-10-27:
@m-mcgowan 
Is device_id in application_dct_t abandon?What is product_store(string format of stm32 CPUID?)?

In 0.5.4 system firmwre
application_dct_t {
      ...
      platform_flash_modules_t flash_modules[MAX_MODULES_SLOT];//100 bytes
      uint16_t product_store[12];
      ....             
}

According to Photon data sheet
https://docs.particle.io/datasheets/photon-datasheet/#memory-map
application_dct_t {
      ...
      platform_flash_modules_t flash_modules[MAX_MODULES_SLOT];//100 bytes
      uint8_t device_id[12];
      ....             
}

m-mcgowan, 2017-11-24:
The device-id in the DCT was [added](https://github.com/spark/firmware/pull/949) in v0.6.0, which is why it isn't present in 0.5.4.  

The part of the DCT before the `application_dct_t` struct is owned by the WICED SDK.

Please use Git issues for reporting problems. For general questions, our [community forum](http://community.particle.io) will be a more suitable place, and you will most likely get answers quicker.

aeris-ming, 2017-11-24:
@m-mcgowan Thanks very much,Is that safe to copy the part of the DCT before the application_dct_t struct from one Photon and flash it to other Photon?

m-mcgowan, 2017-11-24:
You need to copy the parts before application_dct from both sectors, and also duplicate the application_dct_t in both sectors, then it should be ok.

aeris-ming, 2017-11-28:
Thanks @m-mcgowan

closed 2017-11-28 by aeris-ming