particle-iot/device-os #1431: Find out why after call Particle.connect() lost control of RX/TX pin,Is there a Document list all configurations?

aeris-ming, 2017-11-13:
Find Macro define #define USE_SERIAL1 1 in ./system/src/wifitester.cpp
Is there any Doc list all configure about Pins
It is very dispirited after wast days on that issue.
Please Please send me the configure Doc link Thanks!

m-mcgowan, 2017-11-24:
The WiFi tester code is only enabled on entering listening mode, and only if the app has `System.enable(SYSTEM_FLAG_WIFITESTER_OVER_SERIAL1)`.

Could you provide a test application that shows the problem you are seeing?

aeris-ming, 2017-11-24:
Sorry,I made a mistake.

#if Wiring_WiFi
System.enable(SYSTEM_FLAG_WIFITESTER_OVER_SERIAL1);
#endif

I think I do not define Wiring_WiFi,Maybe it is defined in some header.

closed 2017-11-24 by aeris-ming