particle-iot/device-os #1449: Is that possible to change Particle.connect() task to Asynchronous?

aeris-ming, 2017-12-14:
After I move my device out of range of my wifi the led of device blink green.
Screen stop refresh and button stop respond for about 5 minute.
It is very bad if our normal user get into trouble like this.
Is that possible not to block usr thread?

ScruffR, 2017-12-27:
I guess you mean `Particle.connect()` and not `Particle.connected()` as in your `setup()` and there is an existing issue for that already
https://github.com/spark/firmware/issues/1399

aeris-ming, 2018-01-30:
@ScruffR I have read spark#1399 and the patch they already done and I am sure It is not the same issue.
I do not understand why user thread will be block if we are using rt-thread and run usr and system thread in different thread.

ScruffR, 2018-01-30:
As I said above, your code you posted won't show anything useful in connetion with question you asked.
Can you provide some proper snippet?

Also any particular reason for not using a more recent - less "buggy" - system?

aeris-ming, 2018-01-31:
@ScruffR 
No It will block because It has been design to be block!
https://docs.particle.io/reference/firmware/photon/#task-switching
https://docs.particle.io/reference/firmware/photon/#system-functions
For example, when the system is busy connecting to Wi-Fi or establishing the cloud connection and the application calls Particle.variable() then the application will be blocked until the system finished connecting to the cloud (or gives up) so that it is free to service the Particle.variable() function call.

aeris-ming, 2018-01-31:
@ScruffR Can I have your email ?I will send you a video to show how awful it would be.I do not think It is a good idea to publish the video.

aeris-ming, 2018-01-31:
https://github.com/particle-iot/firmware/issues/1392

technobly, 2018-01-31:
The original spec for AUTOMATIC, SEMI_AUTOMATIC and MANUAL is here:
https://github.com/particle-iot/firmware/issues/216

Particle.connect() in SEMI_AUTOMATIC and SINGLE-THREADED mode was always designed to be a blocking call.

ScruffR, 2018-01-31:
Hmm, why then this (as of April 2015 - about a year *after* #216)?
https://community.particle.io/t/create-time-out-for-spark-connect/10351/17?u=scruffr

technobly, 2018-01-31:
"afaict" was what he said there.

Later on he said WiFi.connect() is blocking, wait for multi-threaded.
https://community.particle.io/t/create-time-out-for-spark-connect/10351/40?u=bdub

ScruffR, 2018-01-31:
> "afaict" was what he said there.
And he was right about it back then too.

> Later on he said WiFi.connect() is blocking
And that is only half the story. `WiFi.connect()` is mostly blocking, but *not* completely. From time to time it hands back control to user code (afaict whenever a timeout occures before a new automatic attempt is started). This way user code gets the chance to intervene. Not so with the blocking `Particle.connect()`.

aeris-ming, 2018-01-31:
@ScruffR @technobly WIFI or Internet(tcp or udp whatever) operation code block the user code
Even with System firmware version 0.8.0.rc.1
SYSTEM_MODE(SEMI_AUTOMATIC);
SYSTEM_THREAD(ENABLED);

closed 2018-03-19 by aeris-ming