particle-iot/device-os #1457: IPAddress trap for lazy programmers

ScruffR, 2018-01-09:
`IPAddress` has an overload
```
IPAddress(const uint8_t* address);
```
And `TCPClient` has a `connect()` overload
```
virtual int connect(IPAddress ip, uint16_t port, network_interface_t=0);
```

So one might consider that this should work
```
const uint8_t ipAdr[] = { 10, 0, 0, 8 };
TCPClient tcpClient;
...
  tcpClient.connect(ipAdr, 8000);
```
and it builds without error (and warning), but it doesn't do what's expected.

The (hidden) reason is another `connect()` overload that takes precedence.
```
virtual int connect(const char *host, uint16_t port, network_interface_t=0);
```
Not sure when this overload was added, but as it seems there were some members who got puzzled as of why previously working code didn't anymore.

closed 2020-03-03 by avtolstoy

ScruffR, 2020-03-03:
Was this addressed at all?

avtolstoy, 2020-03-03:
The issue is not reproducible. There is no ambiguity between these two overloads, as `const uint8_t*` cannot be implicitly converted to `const char*`.