particle-iot/device-os #1459: Log - PANIC from Software Timer Callback

hramadan, 2018-01-15:
<details>
  <summary>Have provided source code for a utility called DumpMem() which calls DumpMem16() which highlights an issue with the Log class. 

DumpMem() works when used outside of a Timer Callback, but an SOS PANIC when called within a Timer Callback. The issue is within DumpMem16() and not DumpMem().

The test  code shows that replacing the call to Log.trace() with USBSerial1.printlnf() works fine.
 </summary>

</details>

## Bug Report

### Expected Behavior

Use of the Log class should be equivalent to using *Serial* or *USBSerial1* methods for output, ie if Serial/USBSerial1 works in any given instance, so should Log.

### Observed Behavior

SOS PANIC with 13 flashes == Stack Overflow when the Log output methods are used within a Timer Callback. 

### Steps to Reproduce

Run the Test App with #define CAUSE_PANIC.

Note how the fault is within DumpMem16()

### Test App

```c
/* LOG_TEST_04

Testing v0.8.0.rc1 logging

Reproduce PANIC caused by DumpMem()

*/

/* Version History
2018-01-14 HR
- Created

*/

 #define CAUSE_PANIC

// Set up logger with filters
SerialLogHandler logHandler(LOG_LEVEL_NONE,         // Logging level for non-application messages
    {   // Logging Category Filters
        { "app", LOG_LEVEL_NONE },                  // Default logging level for all application messages
        { "app.network", LOG_LEVEL_TRACE },         // Logging level for networking messages
        { "app.setup", LOG_LEVEL_WARN }             // Logging level for setup messages
    });

void DumpMem(char * pszDesc, uint8_t * pucData, size_t uLen);
void DumpMem16(uint8_t * pucData, size_t uLen, size_t startIdx);

void onTimeout();

// CallbackClass callback;
Timer timer(10000L, onTimeout);

void setup() {

    Serial.begin(9600L);
    USBSerial1.begin(9600L);

    // ==== Dynamically set up logger =====
    
    // Get log manager's instance
    auto logManager = LogManager::instance();

    LogCategoryFilters filters;
    
    filters.append({ "app", LOG_LEVEL_TRACE });

    auto logHandler1 = new StreamLogHandler(USBSerial1, LOG_LEVEL_WARN /* Default level */, filters);

    logManager->addHandler(logHandler1);
  
    delay(2000L);
    
    timer.start(); // starts timer if stopped or resets it if started.
    
    DumpMem("TEST", (uint8_t *) "0123456789",  10);
    Log.trace("Setup exit");
}

void loop() {
    
    static int i = 0;
    
    Log.trace("Loop %d", i);
    onTimeout();                    // <== OK WHEN CALLED FROM HERE
 
    i++;
    
    delay(1000L);
}

void onTimeout()
{
    Log.trace("onTimeout() entry");
    DumpMem("onTimeout()", (uint8_t *) "0123456789",  10);
    Log.trace("onTimeout() exit");
}

//
// Dump data as Hex and Printable ASCII to USBSerial1
//
void DumpMem(char * pszDesc, uint8_t * pucData, size_t uLen)
{
    Log.trace("%s n=%d", pszDesc, uLen);

    // dump in blocks of sixteen
    for (size_t i=0; i < uLen; i= i + 16)
        DumpMem16(pucData, uLen, i);
}

void DumpMem16(uint8_t * pucData, size_t uLen, size_t startIdx)
{
   	char szT[4];
    char szLine[96];
    size_t i;

    
    sprintf(szLine, "  %04X: ", startIdx);

    // print out in Hex
    for (i=startIdx; (i < uLen) && (i < (startIdx + 16)); i++)
    {
        sprintf(szT, "%02X ", (uint8_t) pucData[i]);
        strcat(szLine, szT);
        if (i == startIdx + 7)
            strcat(szLine, "- ");
    }

    // Pad out rest of block
    for (int j = i; j < (startIdx + 16); j++) 
    {
        strcat(szLine, "   ");
        if (j == startIdx + 7)
            strcat(szLine, "- ");
    }

    // output ASCII
    szT[1] = '\0';

    for (i=startIdx; (i < uLen) && (i < (startIdx + 16)); i++)
    {
      	if (((unsigned char) pucData[i] < 0x20) ||
             ((unsigned char) pucData[i] > 0x7F))
            szT[0] = '.';
        else 
            szT[0] = (unsigned char) pucData[i];
            
        strcat(szLine,szT);
    }
    
 #ifdef CAUSE_PANIC
    Log.trace(szLine);              // <== PANICS WHEN CALED FROM upon a timer
 #else
    USBSerial1.printlnf(szLine);      // <== OK
 #endif

} // DumpMem16()

```

### References

https://community.particle.io/t/log-panic-from-software-timer-callback/38672

---

ARONbr1, 2018-01-15:
Quick questions. Why would this print out as Hex and why a print out insted of a log jar.?

ARONbr1, 2018-01-15:
And please correct Me if Im wrong.

hramadan, 2018-01-15:
@ARONbr1, the DumpMem() function outputs something that looks like this:

Test n=32
0120   41 41 41 41 41 41 41 41 - 42 42 42 42 42 42 42 42     AAAAAAAABBBBBBBB
0130   41 41 41 41 41 41 41 41 - 42 42 42 42 42 42 42 42     AAAAAAAABBBBBBBB

It is just a handy way at looking at raw memory.

ARONbr1, 2018-01-15:
Thanks very much. Im bucled up for a ride I guess. I want to learn as much so I can catch up with everyone else.

sergeuz, 2018-01-15:
`Log.trace()` allocates a larger initial buffer on stack than `USBSerial1.printlnf()` (160 vs 20 bytes), and I don't think there's a good general solution here, since a smaller buffer size, e.g. 100 bytes, would still have a potential to cause the stack overflow depending on an application.

As a workaround, I would give a try to the following options:

- Move any stack demanding logic out of the timer callbacks.
- Don't use any functions that do printf-alike formatting in your timer callbacks.
- Use [`Log.print()`](https://github.com/particle-iot/firmware/blob/develop/wiring/inc/spark_wiring_logging.h#L306) instead of `Log.trace()` if your data is already formatted.

hramadan, 2018-01-15:
@sergeuz, that is a very helpful analysis and it all makes sense! 

I will rejig the DumpMem16() routine using Log.print() and report back (may be a a day or so from now).

sergeuz, 2018-01-15:
@hramadan I would recommend to avoid doing any logging in the timer callback though, at least in production. Writing to a serial interface is a blocking operation and it may affect all other software timers, since they are processed in the same RTOS thread.

technobly, 2018-01-16:
Setting flags in Timer callbacks to be later processed in loop() is the best design pattern I can recommend here.

hramadan, 2018-01-16:
@technobly, @sergeuz, my DumpMem routine has caused panics **outside** of a timer callback, and that is the issue that needs solving. Problem is that it is hard to reproduce.

My thinking is that if we solve the issue within the supplied test app (ie inside a timer callback), then it will probably solve the other problem too. 

Just need to find the time to do the test as suggested by @sergeuz!

hramadan, 2018-01-20:
@sergeuz, @technobly pleased to announce that using **Log.print()** instead of **Log.trace()** solved the issue - the test app worked without panic and the original issue - panic from a routine called with setup() was also cleared. 

Thanks @sergeuz!

closed 2018-01-20 by hramadan