particle-iot/device-os #1461: Photon TCP can blocks indefinitely with new wiced 3.7 (release 0.7x and 0.8x)

tlangmo, 2018-01-15:
## Bug Report
Whenever actively sending over tcp, the application thread can stall indefinitely long in the new releases. This causes the application watchdog to timeout and generally stops any use code from executing.

### Expected Behavior
On Wiced SDK 3.3 (particle release 0.6.x and prior), sending tcp data returned WICED_RESULT_WOULD_BLOCK (-16) whenever the network stack was not able to send the data. This typically happens when the connection quality is poor, for example if the device moves out of the wifi range. The calling thread continued executing and the user code can handle this condition properly (such as sending at a later point)

### Observed Behavior
In Wiced SDK 3.7, this behavior was changed according to its changelog.txt.
`LwIP network_tcp_send_packet() no longer sets NETCONN_DONTBLOCK when calling netconn_write()`

Now, TCPClient::write and everything else which uses wiced_tcp_send_buffer will simply block.
This causes user code to stall whenever having a bad wifi signal during an active tcp connection. I have seen stalls of many minutes until either the Wifi connection reset itself, the photon crashed or the photon resets itself, similar to watchdog (although I have not activated one)

### Steps to Reproduce
The key is to have a weak WiFi signal. This can be achieved by walking out of range or shielding the signal. Using the external antenna and unplugging it worked in many cases as well. 

Compiling against releases < 0.7.x shows that the app loop never stalls

### Test App  (application.cpp)
The app monitors the app thread by blinking and serial output every 2 seconds. 
1. Connect to the photon via the python script
2. Reduce wifi signal and perceive the blocking

```c
#include "application.h" 

PRODUCT_ID(PLATFORM_ID);

SYSTEM_THREAD(ENABLED);
SYSTEM_MODE(SEMI_AUTOMATIC);
STARTUP(WiFi.selectAntenna(ANT_EXTERNAL));
STARTUP(System.enableFeature(FEATURE_RESET_INFO));

SerialLogHandler logHandler(LOG_LEVEL_INFO);

#define WIFI_SSID  "ssid"
#define WIFI_PWD  "password"

TCPClient tcpConn;
TCPServer  tcpServer(5000);

void destroyNetworkResources() {
	Log.info("destroyNetworkResources");
	tcpServer.stop();
	tcpConn.stop();
}
void createNetworkResources() {
	Log.info("createNetworkResources");
	tcpServer.begin();
}

void handler_event_network_status(system_event_t event, int data, void* param)
{
	if (event == network_status) {
		Log.info("handler_event_network_status %d", data);
		switch (data) {
		// make sure before we mess with sockets etc. our resources are destroyed
		case network_status_disconnecting:
			destroyNetworkResources(); break;
		case network_status_connected:
			createNetworkResources(); break;
		default:;
		};
	}
}

uint32_t now = 0;

/* This function is called once at start up ----------------------------------*/
void setup()
{
	WiFi.on();
	WiFi.useDynamicIP();
	// clear credentials via buttons to set the hardcoded credentials
	if (!WiFi.hasCredentials()) {
		WiFi.setCredentials(WIFI_SSID, WIFI_PWD, WPA2, WLAN_CIPHER_AES_TKIP);
	}
	System.on(network_status, handler_event_network_status);
	WiFi.connect(WIFI_CONNECT_SKIP_LISTEN);
}

int lastError = 0;
/* This function loops forever --------------------------------------------*/
void loop()
{
    //This will run in a loop
	if (int(millis()) - int(now) > 2000) {
		RGB.control(true);
		RGB.color(0,0,255);
		delay(100);
		RGB.control(false);
		now = millis();
		Log.info("%s",WiFi.localIP().toString().c_str());
	}
	TCPClient newClient = tcpServer.available();
	if (newClient.connected()) {
		tcpConn = newClient;
	}
	if (tcpConn.connected()) {
		uint8_t buf[500];
		memset(buf,0xee, 500);
		int rc = tcpConn.write(buf,sizeof(buf));
		if (rc < 0) {
			if (rc != lastError) {
				Log.info("TcpClient::write returned rc=%d (WOULD_BLOCK==-16)", rc);
				lastError = rc;
			}
		}
	}
	else {
		tcpConn.stop();
	}

}
```
### Python tcp client (tcp_test.py)
```
import socket

s = socket.socket(socket.AF_INET, socket.SOCK_STREAM)
s.connect(("10.39.105.215", 5000))
total_received = 0
print("connected")
while True:
    data = s.recv(1000)
    total_received += len(data)
    print("Received: {:.1f} KB        ".format(total_received/1024), end="\r")
```

zacnelson, 2018-01-16:
I am also experiencing this behavior on a Photon running v0.8.0-rc.1. I added system level logging `SerialLogHandler logHandler(LOG_LEVEL_ALL);` but the system firmware doesn't appear to send any logs during this event. Will continue to test and gather more information.

sirhcil, 2018-07-31:
I have also experienced our device freezing and the symptoms sound similar. Spotty Wi-Fi, non-responsive user code. Used 0.7.0, 0.8.0rc4 and both exhibited this behavior. Are you guys using system threading?

Bacichetti, 2018-07-31:
I believe this issue is the same been reported at the forum here:
[Issues with firmware 0.7.0](https://community.particle.io/t/issues-with-firmware-0-7-0/41808)

However, some differences reported indicate the problem is broader:
1) the same behaviour happens with the TCP/UDP socket used by the system firmware to keep checking on Cloud connection (if I'm not wrong).
2) reports from the community indicate that the problem happens regardless of the WiFi quality given enough running time.

avtolstoy, 2020-03-03:
This issue should be resolve in recent releases.

closed 2020-03-03 by avtolstoy